VUnit/vunit #128: paths/folders generated by vunit are not recognised as valid by modelsim

swattor, 2016-02-19:
The default format for a testcase results folder - e.g.

vunit_out/tests/lib.testbench.PARAMETER=X.test1/modelsim/vsim.wlf

isn't valid for modelsim/questa which means you have to navigate to that folder to open it - otherwise you end up with an error "Dataset names can only contain alphanumeric characters or underscores."

Given that a lot of programs tend to use a limited char set for path/filenames perhaps this should be made more friendly?

kraigher, 2016-02-21:
@swattor When do you get that error? I have used similar path names myself with ModelSim and it worked just fine.

kraigher, 2016-03-13:
@swattor Is this issue still relevant?

LarsAsplund, 2016-06-09:
@swattor Is this issue still relevant?

LarsAsplund, 2016-06-15:
Anyone with access to Questa that has seen this or can recreate this? If not, I will close this issue.

closed 2016-06-26 by LarsAsplund