VUnit/vunit #130: Question/Doc enhancement: Bringing VUnit runner functionality to existing testbenches

joshrsmith, 2016-02-20:
How to use VUnit for completely new testbenches is pretty well documented, starting with the most basic cases shown in the user guide.

I think a common questions for someone new to VUnit is: _What do I need to do to adapt existing/legacy testbenches to utilize VUnit?_

Developing an automated testbench is good practice, so it is safe to assume that users out there already have test code that was written without VUnit in mind. Without VUnit, a common practice would probably be to lump a variety of tests into a single testbench for a single entity. The testbench is then run and failures are reported via assertions, and the testbench is automatically stopped if the severity is high enough. If there are no failures, the testbench would end automatically (after a fixed duration, or with a call to `std.env.finish`).

Assuming someone has a set of existing testbenches that are written in such a manner, what is the path of least resistance to adopting the VUnit python running framework (automatic compilation, test detection, parallel simulation, reporting results via XML)?  Obviously some of the benefits of the VUnit VHDL test library will be missed, but for legacy code this is acceptable -- the testbench is already complete.

LarsAsplund, 2016-02-20:
I can add more information on this but basically you would take the code in your legacy main process and put that in the place of
``` vhdl
report "Hello World!";
```
in the first example [here](http://vunit.github.io/user_guide.html#vhdl-test-benches). Legacy calls to `std.env.finish` are removed.
``` vhdl 
test_runner_cleanup(runner);
```
takes care of that.

joshrsmith, 2016-02-21:
So far as I understand:

- Add in library references to VUnit library (as in examples)
- Make sure to have `runner_cfg` in the generic map (because this is what the scanner uses to identify a testbench?)
- Add `test_runner_setup` call - at beginning of main process.
- Replace call to `std.env.finish` with `test_runner_cleanup`

In essence, use the template of the user guide, replacing "hello world" with what is actually being done to test stuff.

Follow up -- Is it possible to use VUnit as a test runner without the VUnit library at all? I suppose there would need to be a way to manually indicate to VUnit where the testbenches are, since the built-in scanning functionality would not work.

kraigher, 2016-02-21:
@joshrsmith Yes it is in theory possible to use VUnit as a test runner without the VUnit library. It is not something that we have support for using the Public API. The private classes of VUnit can be combined to do it without much effort though. My opinion is that it is so easy to convert an existing test bench to VUnit, just add test_runner_setup/test_runner_cleanup to your main test bench process, that such support is not necessary. It would probably be less work to do that than to describe to VUnit how the test bench should run and what is the pass/fail criteria. Do you agree?

joshrsmith, 2016-02-21:
@kraigher I agree that it is easy to make the conversion to a VUnit-compatible testbench. My concern would but that by doing so, that running the testbench (with or without the VUnit runner) now requires the VUnit library. 

When using the VUnit runner approach, this is not a problem because VUnit library is automatically included in the compilation of dependencies. However, if running a testbench via simulator directly (without VUnit runner), the users must first compile the VUnit library sources into an appropriately named library and ensure it is mapped into their project.  This isn't insurmountable via training, but it not a "transparent"/backwards compatible change to add VUnit support to existing testbenches.  For example, if I were to update existing testbench code of widely shared modules so that they could be tested via VUnit runner, now anyone else that uses those shared modules must also learn where to get VUnit sources, compile, and map. If they didn't realize they needed to do this, all of the testbenches would now appear broken.

In my experience, managing 3rd party libraries for VHDL is a painful setup step -- it is just a lot easier whenever user has the libraries by default (e.g. those bundled with simulator, _ieee, osvvm_ and some vendor specific libraries).

That said, I don't know if there is a good solution for VUnit to avoid this - perhaps it is just the price of adoption.

kraigher, 2016-02-21:
@joshrsmith 
Using a testing framework is always a dependency. In the software world people are not afraid of depending on testing frameworks for increased productivity. For many languages there is a de-facto standard framework to use for testing. If a project chooses to use VUnit then it will have to accept that it is a dependency. VUnit will make it easier for multiple designers to collaborate on shared modules since it unifies things that might otherwise be fragmented. VUnit also makes it easy to quickly use other libraries in your test benches. In commercial projects where I have used VUnit it has been either a git submodule in the project repository or checked in along with the code.

If you for some reason do wish to have a shallow dependency on VUnit you can refrain from using anything more than test_runner_setup/test_runner_cleanup and then provide dummy implementations of those in a single package that can be used to run the test bench outside of VUnit.

LarsAsplund, 2016-02-21:
@joshrsmith If you need the ability to run a VUnit testbench within a simulator without the Python runner you should also include a default value for the `runner_cfg` generic.

``` vhdl
entity tb_example is
  generic (runner_cfg : runner_cfg_t  := runner_cfg_default);
end entity;
```

This default will run all test cases in your testbench. Be aware that they will run in the same simulation so there is a risk of dependencies causing test cases to fail although the problem is a side effect from a previously run test case.

cmarqu, 2016-02-21:
This is a very interesting problem, and I can absolutely see where @joshrsmith is coming from. 
"If a project chooses to use VUnit then it will have to accept that it is a dependency." - that is not quite so clear cut in my experience. There usually is someone who sees that a new tool/library/package/etc. is a good idea, and now has to either fully "sell" it to other persons with the information from the new tool's feature sheet (then leading to an official evaluation maybe), or start small and show the benefits in an existing project. With the latter, it helps when a key feature can be introduced without disrupting the work flow at all for other people (i.e. when I can add it to a project in a self-contained manner - tell colleagues "just update everything and the project is consistent again, and then take a look at 'foo' - isn't that cool?").
I think a documentation section like "Adopter's Guide" would be useful, and I have taken some notes over the last weeks that I'll add tomorrow.

LarsAsplund, 2016-02-21:
@cmarqu Thanks, such input is appreciated.

cmarqu, 2016-02-22:
Here are the points which were relevant in my setup (please review thoroughly, I might have introduced inaccuracies):
- VUnit derives the compilation order from the list of files it is given, which thus should be complete. It then compiles each of these files separately, which means that each file needs to be self-contained, i.e. in the case of Verilog files, needs to have all used defines set. Defines that are set from the simulator need to be defined with the ``defines`` argument of ``add_source_file(s)`` so that VUnit knows about them.
- If your existing tests are not placed in the toplevel testbench module, you can set ``tb.scan_tests_from_file()``. In addition, you must propagate the ``runner_cfg`` generic/parameter through the hierarchy from the toplevel testbench to the entity/module containing the tests.
- Your testbench must not stop the simulation before ``test_runner_cleanup()`` (for VHDL) or ``TEST_SUITE_CLEANUP`` (for Verilog) because otherwise the correct simulator exit status will not be set.

LarsAsplund, 2016-06-09:
@joshrsmith Are you planning to extend your proposal for an adopters guide? If not,we should proceed with a pull request and see if we can get this issue closed.

joshrsmith, 2016-06-09:
I think I just want to abandon this at this point.

LarsAsplund, 2016-06-10:
@joshrsmith I've added a few lines to the user guide containing the essence of what have been discussed here such that this issue can be closed.

closed 2016-06-10 by LarsAsplund