VUnit/vunit #139: Riviera 2015.10 does not pass acceptance test

joshrsmith, 2016-03-01:
I recognize this is not on the officially supported simulator list, but I thought I would give it a try for compatibility checking.

Unfortunately the error message from Riviera is not helpful at all.

```
test_com_debug_vhdl_2008 (test_external_run_scripts.TestExternalRunScripts) ... Compiling e:\vunit\vunit\vhdl\vhdl\src\l
ib\std\textio.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\vhdl\src\lang\lang.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\string_ops\src\string_ops.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\stop_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\stop_body_2008.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\path\src\path.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log_types.vhd into vunit_lib ...
COMP96 WARNING COMP96_0119: "The range is 'null range'." "e:\vunit\vunit\vhdl\logging\src\log_types.vhd" 26 55
Compiling e:\vunit\vunit\vhdl\logging\src\log_formatting.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log_special_types200x.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log_base_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log_base.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\logging\src\log.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_types.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check_types.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check_special_types200x.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check_base_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check_base.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\check\src\check.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\dictionary\src\dictionary.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run_types.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run_special_types200x.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run_base_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run_base.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\run\src\run.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\vunit_run_context.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\vunit_context.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_std_codec_builder.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_debug_codec_builder.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_string.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_codec_api.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_codec_debug.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com.vhd into vunit_lib ...
Compiling e:\vunit\vunit\vhdl\com\src\com_context.vhd into vunit_lib ...
Compiling E:\vunit\vunit\vhdl\com\test\tb_com.vhd into tb_com_lib ...
Compiling E:\vunit\vunit\vhdl\com\test\more_constants.vhd into tb_com_lib ...
Compiling E:\vunit\vunit\vhdl\com\test\constants.vhd into tb_com_lib ...
Compiling E:\vunit\vunit\vhdl\com\test\custom_types.vhd into tb_com_lib ...
Compiling E:\vunit\vunit\test\acceptance\external_run_out\codecs\tb_com_lib\custom_codec_pkg.vhd into tb_com_lib ...
DAGGEN ERROR DAGGEN_0700: "Fatal error : INTERNAL CODE GENERATOR ERROR
 (during analysis of model custom_codec_pkg/custom_codec_pkg)"
COMP96 Compile failure 1 Errors 0 Warnings  Analysis time :  47.0 [ms]
FAIL

======================================================================
FAIL: test_com_debug_vhdl_2008 (test_external_run_scripts.TestExternalRunScripts)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "E:\vunit\vunit\test\acceptance\test_external_run_scripts.py", line 177, in test_com_debug_vhdl_2008
    args=["--use-debug-codecs"])
  File "E:\vunit\vunit\test\acceptance\test_external_run_scripts.py", line 195, in check
    self.assertEqual(retcode, exit_code)
AssertionError: 1 != 0

----------------------------------------------------------------------
Ran 5 tests in 37.601s

FAILED (failures=1, skipped=2)
```

LarsAsplund, 2016-03-01:
The trial license we had to prove that it was possible to use VUnit with Riviera-PRO and Active-HDL has expired so we're in the process of getting our Unite partnership in place for a more long-term solution. Unfortunately this has left us with a period where we can't verify new releases. The `com` package is by far the most challenging package for any simulator so problems in that area isn't really a surprise. The internal error that you're seeing is something that I've experienced before and the I solved it by removing functionality until the problem was isolated.

joshrsmith, 2016-03-10:
I just verified that 2016.02 has the same issue.

LarsAsplund, 2016-03-10:
Ok. we're closer to getting our licenses but still not there. Will address this when I have them.

joshrsmith, 2016-03-12:
Far as I can tell, this is the offending line in `custom_codec_pkg.vhd`

```
  function decode (
    constant code : string)
    return array7_t is
-- ...
    variable ret_val : array7_t(ret_val_range(code)'range(1), ret_val_range(code)'range(2));
-- ...
```

LarsAsplund, 2016-03-13:
@joshrsmith This seems like the same problem I fixed some time ago but for the debug codecs. At that time I was confused that I only had to add the workaround to the standard codecs. Test the aldec-com-debug branch

joshrsmith, 2016-03-13:
Far as I know this only happens when the flag to use debug codecs is specified. Is that what you mean? Does that make sense with what I posted?

LarsAsplund, 2016-03-13:
@joshrsmith Yes, the failing codec is only genetated with the `--use-debug-codecs` option. The workaround I added now is the same I've added previously for the codec generated  without this option. Could you give it a try?

joshrsmith, 2016-03-13:
Now it will compile, but it does not pass. Basically all of the encode/decode checks fail, plus a couple others, when running tb_com_lib.tb_com_codec.* with `--use-debug-codecs`.

Not sure if this is new, but in the log I see:
`#KERNEL: Error: Unable to allocate 2048.00 MB VHDL dynamic memory.`

I could try with a 64-bit install w/64-bit license, but even if it does work that doesn't make for a great long-term solution.

Everything passes without specifying `--use-debug-codecs` still.

LarsAsplund, 2016-03-13:
@joshrsmith Haven't seen that one before. Anyway, I just recalled that the test for array7 was removed from the testbench since I couldn't make it work with Active-HDL when we first launched support for that simulator. I think this is acceptable considering the rare use case.

``` vhdl
  type fruit_t is (apple, banana, melon, kiwi, orange, papaya);
  type array7_t is array (integer range <>, fruit_t range <>) of natural;
``` 
So there's really no point in keeping the type definition and make codecs for it. Try my latest push and see if it works when it has been removed. 

There is still a test for a more common unconstrained 2D array

``` vhdl
  type array5_t is array (integer range <>, integer range <>) of natural;
```

joshrsmith, 2016-03-13:
Alright, so it turns out 2016.02 _still_ has the same kernel issue I mentioned before, but every other version (9.3 - 2015.10) passes the acceptance test. So it is possible that the fix you made before removing array7 completely was "ok" but something is more seriously broken in 2016.02.

Perhaps when you get a license setup you can submit a more concise test case? I would be willing to submit one, but I would have a tough time boiling this down to the most simple case to reproduce.

LarsAsplund, 2016-03-13:
Ok, but at least we've verified that it works with 2015.10 which is one version newer than what has been confirmed before. Did you test both 32- and 64-bit versions of 2015.10? If so, I can merge and close this issue and open a new for 2016.02.

joshrsmith, 2016-03-13:
I do not have 64-bit versions installed. I can work towards getting that up this week sometime.

joshrsmith, 2016-03-14:
Passing:
ActiveHDL - 9.3, 10.1 (x86/x64), 10.2 (x86/x64)
Riviera- 2015.06 (x86/x64), 2015.10 (x86/x64)

Failing:
Riviera- 2016.02 (x86/x64)  - will make a new issue for this

LarsAsplund, 2016-03-14:
Thanks. I've merged the fix to master.

closed 2016-03-14 by LarsAsplund