VUnit/vunit #144: Cryptic vhdl_parser.py error

adrspa, 2016-03-09:
This is the traceback I got:

Traceback (most recent call last):
  File "vunit/vunit/project.py", line 699, in parse
    design_file = parser.parse(code, self.name, self._content_hash)
  File "vunit/vunit/vhdl_parser.py", line 60, in parse
    design_file = VHDLDesignFile.parse(code)
  File "vunit/vunit/vhdl_parser.py", line 93, in parse
    return cls(entities=list(VHDLEntity.find(code)),
  File "vunit/vunit/vhdl_parser.py", line 324, in find
    yield VHDLEntity.parse(sub_code[:match.end()])
  File "vunit/vunit/vhdl_parser.py", line 343, in parse
    generics = cls._find_generic_clause(code)
  File "vunit/vunit/vhdl_parser.py", line 370, in _find_generic_clause
    code[match.start(): match.end() + closing_pos + match_semicolon.end()])
  File "vunit/vunit/vhdl_parser.py", line 455, in _parse_generic_clause
    generic_list.append(VHDLInterfaceElement.parse(interface_element))
  File "vunit/vunit/vhdl_parser.py", line 573, in parse
    mode_split = interface_element_string.split(':')[1].strip().split(None, 1)
IndexError: list index out of range
  ERROR - Failed to parse /vobs/rbs/hw/ru_fpga/ru_lib/ti/tb/tb_ti_str.vhd
Traceback (most recent call last):
  File "run.py", line 91, in <module>
    tb_ti_str = lib.entity("tb_ti_str")
  File "vunit/vunit/ui.py", line 991, in entity
    raise KeyError(name)
KeyError: 'tb_ti_str'

The extra ";" in the entity made it crash. A more intelligible error message would be nice. :-)

ENTITY tb_ti_str IS
  GENERIC (runner_cfg            : runner_cfg_t;
           g_test_clock_en_speed : INTEGER;
           g_axi_stim_file       : STRING;
           g_data_0_file         : STRING := "";
           g_data_1_file         : STRING := "";
           );
END tb_ti_str;

kraigher, 2016-03-09:
The current VHDL parser is a bit rudimentary and one of the oldest parts of VUnit. 
It is pretty much based on loose regexps and it does not have much error handling or good error reporting. I have plans to re-write the VHDL parser based on tokenization with location information in the same way as the current Verilog parser. After this work is finished the error messages will be much better.

adrspa, 2016-03-11:
That sounds good! Just thought I should report it in so you guys can improve the tool.

LarsAsplund, 2016-06-10:
All issues that can be solved with a more advanced VHDL parser will be tracked in #174 .

closed 2016-06-10 by LarsAsplund