VUnit/vunit #147: add_external_library only available at project level

joshrsmith, 2016-03-11:
I see that `add_external_library` is only available at the project level. How should one handle the case where different libraries that are named the same thing may be needed for different components within the project?

For example: Xilinx `unisim` and Aldec. Aldec supplies two pre-compiled unisim libraries, one for vivado and one for ISE.  If `moduleA` depends on the unisim for vivado, and `moduleB` depends on the unisim for ISE, it seems impossible to do this in the same project.

I hate that this is an issue at all -- I wish Aldec supplied everything wrapped into a single library, but perhaps the issue goes deeper than that.

Are there any similar cases with other vendors or silicon vendors?

One possible solution to this case is to allow setting the external library to be scoped similar to other configurations. Then you could set it at a global level (the default), and possibly override at the library/testbench/test case level.

kraigher, 2016-03-12:
I do not think it is even possible, from the simulator point of view, to mix modules under the same top level requiring different versions of a library. For a simulation there will always be a library.cfg or modelsim.ini that uniquely maps library names to library locations. Thus the scenario you describe means that there are two disjoint set of files where the ones that uses the ISE version cannot depend on files using the Vivado version and vice versa. My recommendation would thus be to not mix modules requiring a different version of an external library in the same run.py but have two run.py files such as run_vivado.py and run_ise.py. You can still share code between two run.py files since it is just Python.
Would that work for you?

To support this in VUnit we would have to add multiple compile contexts where one file from one compile context cannot depend on a file in another compile context but maybe on files in a parent compile context. This is similar to having multiple run.py files but the testcases could run in the same suite and there could be common dependencies between the sets of files. This is not a quick job and probably not something that we should do if this is not a very common problem. I think people will use either Vivado or ISE in their project not both.

joshrsmith, 2016-03-12:
Yes, this is messy all around. I agree with your points, and this is likely a far our edge case that can be handled by multiple projects (and is probably clearer that way).

closed 2016-03-12 by joshrsmith