VUnit/vunit #148: ActiveHDL 10.2 not detected

joshrsmith, 2016-03-12:
It appears in Active HDL 9.3, there is `vsimsa.bat` and `VSimSA.exe`, but in 10.2 it changed to `vsimsa.bat` and `vsimsa.exe`.

When running VUnit against ActiveHDL 10.2 it does not detect the simulator. It looks for VSimSA (case sensitive), which it does not find.

As a possible fix, I changed the search to use `vsimsa`, which allows ActiveHDL to be detected, but this then causes confusion between RivieraPRO and ActiveHDL since they use the same method of detection (e.g. even if VUNIT_SIMULATOR=rivierapro then ActiveHDL will be used if first on the path).

One difference between activehdl and riviera is that activehdl bin directory has `avhdl.exe` and riviera has `riviera.exe` (this is on windows -- not sure how it looks on Linux). However, even if the two simulators can be successfully detected and differentiated, then an absolute path will be needed to invoke `vcom` and `vsim`, because otherwise the order in the system path will take precedent.

Related questions:

- Does VUnit want to support Active HDL 9.x? (user guide only claims support for 10.2). 9.x is getting pretty old now. I personally wouldn't mind not having it supported.
- How can the detection / differentiation between Active HDL and RivieraPRO be improved?
- Should VUnit provide a way to explicitly specify a path to simulator to use in cases where multiple versions of the same simulator are on the path?

joshrsmith, 2016-03-12:
This worked on windows, allowing me to use `VUNIT_SIMULATOR` to select between tools when they were both on the path.

```
    def _find_prefix(cls):
        return cls.find_toolchain(["vsim",
                                   "vsimsa",
                                   "riviera"])
```

and

```
    def _find_prefix(cls):
        return cls.find_toolchain(["vsim",
                                   "vsimsa",
                                   "avhdl"])
```

kraigher, 2016-03-12:
I think we should listen to a a <SIMULATOR_NAME>_PATH environment variable to let the user explicitly set the executable PATH in any case.

joshrsmith, 2016-03-12:
One thing which you can possibly take advantage of is that there is no Active HDL for Linux: https://www.aldec.com/en/support/resources/documentation/faq/1742
http://www.eetimes.com/document.asp?doc_id=1215213

kraigher, 2016-03-12:
@joshrsmith Ok based on this information the rule can be:
rivierapro = has(vsim) && has(vsimsa) && not_has(avhdl)
activehdl = has(vsim) && has(avhdl)

joshrsmith, 2016-03-12:
Great! This appears to be working. Versions 9.3, 10.1, 10.2 are being correctly detected, as are Riviera versions. Active HDL versions 9.3, 10.1 and 10.2 are all passing acceptance tests as well.

kraigher, 2016-03-12:
Good news. Thank you!

commit 63741c79 (2016-03-13): Added possibility to set <SIMULATOR_NAME>_PATH environment variable to explicitly set simulator executable prefix. Closes #148
commit 7d3836db (2016-03-13): Added possibility to set VUNIT_<SIMULATOR_NAME>_PATH environment variable to explicitly set simulator executable prefix. Closes #148
closed 2016-03-13 by kraigher