VUnit/vunit #161: Running VUnit with ModelSim DE 10.1a results in SIGSEGV

suoto, 2016-04-15:
I'm running the example from [examples/vhdl/uart] (https://github.com/VUnit/vunit/tree/master/examples/vhdl/uart) folder and ModelSim DE 10.1a is terminating with SIGSEGV when launching the vsim command.

I have both ModelSim DE 10.1a and ModelSim Altera 10.3c and this happens **only** with ModelSim DE 10.1a. (I have cleaned all generated files when switching between the two).

The shell output is as follows:

```
~/dev/vunit/examples/vhdl/uart $ python run.py
Compiling src/test/uart_model_pkg.vhd into tb_uart_lib ...
Compiling src/uart_tx.vhd into uart_lib ...
Compiling src/uart_rx.vhd into uart_lib ...
Compiling src/test/tb_uart_tx.vhd into tb_uart_lib ...
###### /home/asouto/dev/vunit/examples/vhdl/uart/src/test/tb_uart_tx.vhd(111):   end process;
** Warning: [2] /home/asouto/dev/vunit/examples/vhdl/uart/src/test/tb_uart_tx.vhd(111): (vcom-1090) Possible infinite loop: Process contains no WAIT statement.
Compiling src/test/tb_uart_rx.vhd into tb_uart_lib ...
Starting tb_uart_lib.tb_uart_rx.test_tvalid_low_at_start
# vsim -L uart_lib -L vunit_lib -L tb_uart_lib -L osvvm -modelsimini /home/asouto/dev/vunit/examples/vhdl/uart/vunit_out/modelsim/modelsim.ini -onfinish stop -quiet -t ps -wlf /home/asouto/dev/vunit/exam
ples/vhdl/uart/vunit_out/tests/tb_uart_lib.tb_uart_rx.test_tvalid_low_at_start/modelsim/vsim.wlf {-g/tb_uart_rx/runner_cfg="enabled_test_cases : test_tvalid_low_at_start,output path : /home/asouto/dev/vu
nit/examples/vhdl/uart/vunit_out/tests/tb_uart_lib.tb_uart_rx.test_tvalid_low_at_start/,active python runner : true"} tb_uart_lib.tb_uart_rx(tb) 
# //  ModelSim DE 10.1a Feb 21 2012 Linux 3.16.0-0.bpo.4-amd64
# //
# //  Copyright 1991-2012 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Fri Apr 15 10:27:48 2016
# ModelSim Stack Trace
# Program = vsim
# Id = "10.1a"
# Version = "2012.02"
# Date = "Feb 21 2012"
# Platform = linuxpe
# 0    0xf6e7176d: '/home/asouto/.local/lib/python2.7/site-packages/vunit/vhdl/string_ops/src/string_ops.vhd:315'
# 1    0xf6da16d1: '/home/asouto/.local/lib/python2.7/site-packages/vunit/vhdl/string_ops/src/string_ops.vhd:212'
# 2    0x0817c48d: '<unknown (@0x817c48d)>'
# 3    0x082bb64e: '<unknown (@0x82bb64e)>'
# 4    0x082bbe1a: '<unknown (@0x82bbe1a)>'
# 5    0x0833231f: '<unknown (@0x833231f)>'
# 6    0x083337cd: '<unknown (@0x83337cd)>'
# 7    0x08627034: '<unknown (@0x8627034)>'
# 8    0x08984cc7: '<unknown (@0x8984cc7)>'
# 9    0x08986479: '<unknown (@0x8986479)>'
# 10   0x089abdab: '<unknown (@0x89abdab)>'
# 11   0x089b0e93: '<unknown (@0x89b0e93)>'
# 12   0x0898840c: '<unknown (@0x898840c)>'
# 13   0x0898b4bd: '<unknown (@0x898b4bd)>'
# 14   0x08986479: '<unknown (@0x8986479)>'
# 15   0x089abdab: '<unknown (@0x89abdab)>'
# 16   0x089b10f3: '<unknown (@0x89b10f3)>'
# 17   0x08986b8c: '<unknown (@0x8986b8c)>'
# 18   0x0898e28d: '<unknown (@0x898e28d)>'
# 19   0x08986479: '<unknown (@0x8986479)>'
# 20   0x0898790d: '<unknown (@0x898790d)>'
# 21   0x08987d50: '<unknown (@0x8987d50)>'
# 22   0x0898873a: '<unknown (@0x898873a)>'
# 23   0x08750e23: '<unknown (@0x8750e23)>'
# 24   0x089beaab: '<unknown (@0x89beaab)>'
# 25   0x089ea622: '<unknown (@0x89ea622)>'
# 26   0x089cc930: '<unknown (@0x89cc930)>'
# 27   0x089ccc2e: '<unknown (@0x89ccc2e)>'
# 28   0x088f2b00: '<unknown (@0x88f2b00)>'
# 29   0x08448713: '<unknown (@0x8448713)>'
# 30   0x08379305: '<unknown (@0x8379305)>'
# 31   0x080603c6: '<unknown (@0x80603c6)>'
# End of Stack Trace

** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
(Exit codes are defined in the ModelSim messages appendix
of the ModelSim User's Manual.)
fail (P=0 S=0 F=1 T=6) tb_uart_lib.tb_uart_rx.test_tvalid_low_at_start (1.5 seconds)
```

For the record, the full ModelSim version strings are

```
Model Technology ModelSim DE vcom 10.1a Compiler 2012.02 Feb 21 2012
Model Technology ModelSim ALTERA vcom 10.3c Compiler 2014.09 Sep 20 2014
```

I'm using Debian 7.10 and VUnit 0.52.0.

If you need more info please let me know.

LarsAsplund, 2016-04-15:
@suoto Unfortunately we don't have the type of tool provider deal with Mentor like we do with Aldec so we don't have access to any DE licenses. Anyway, I'm noticing that you are having problems with one of the basic examples. Is this the first thing you've tested or do you see the same problems for other examples as well? Also, your DE license is a few years old. Do you see the same problem with an equally old Altera edition? That we could debug.

suoto, 2016-04-15:
I just began using VUnit actually (could be my fault somewhere...). I had success with the Altera's version testing other VUnit examples and some code at work also.
I haven't tested with other ModelSim versions, I can try to test with an older Altera version and check if the same thing happens.

LarsAsplund, 2016-04-15:
I would expect that this has to do with the tool rather than a mistake from your side. It not much that you can do wrong when trying one of the examples, especially when it works with AE. Let's see what your investigation on an older AE reveals and take it from there.

kraigher, 2016-04-15:
The problem is with very likely that Modelsim DE 10.1a faults on some part of the VUnit VHDL code. So it is not something you have done wrong. Maybe this part of the code can be rewritten but we have no way of trying it ourselves. If the error could be duplicated in some altera modelsim version maybe we could.

suoto, 2016-04-15:
I've tested with other versions and the results are as follows:

| Version                                 | Result                                                   |
 ---------------------------------------- | -------------------------------------------------------- |
| ModelSim ALTERA 6.6d (11/02/2010)  | Unknown, can't compile AlertLogPkg.vhd                   |
| ModelSim ALTERA 10.0c (09/21/2011) | Unknown, can't compile AlertLogPkg.vhd (different error) |
| ModelSim ALTERA 10.1b (04/27/2012) | Everything runs fine                                     |
| ModelSim ALTERA 10.3c (09/20/2014) | Everything runs fine                                     |
| ModelSim ALTERA 10.4b (05/27/2015) | Everything runs fine                                     |

Unfortunately I couldn't find an Altera's 10.1**a** version to try.

LarsAsplund, 2016-04-15:
Thanks. It turns out that I had old Altera editions installed, both 10.0d and 10.1b. 10.0d fails on AlertLogPkg.vhd (part of OSVVM) as well so I tested the user guide examples which doesn't include OSVVM. Then it fails with
```
Context declaration not yet implemented.
```
OSVVM is also VHDL-2008 so this is probably just a case of lacking language support.
10.1b is working just fine and I suspect that our claims of supporting 10.1 - 10.4 should be 10.1b - 10.4 since ModelSim AE is what we've been using for verification. Anyway, it might be possible for me to get hold of the correct version with ModelSim PE and run the tests with that.

LarsAsplund, 2016-04-20:
Fixed by #163

closed 2016-04-20 by LarsAsplund