VUnit/vunit #164: Aldec Riviera-PRO parallel simulations fail

LukasVik, 2016-04-20:
The core problem is that riviera uses the compiled libraries folder as a working directory. Running with  --num-threads greater than one will fail if it ever runs more than one testbench from a tb library at the same time.

Say for example that I run a simulation with args: tb_horse_lib.* --num-threads 4.
One simulation will start just fine, using vunit_out/rivierapro/libraries/tb_horse_lib/ as a temp folder. The other three will stop, saying that the module tb_horse_lib is locked.

I have a workaround available that copies the libraries folder to the test case output folder: https://github.com/LukasVik/vunit/commit/f9a9cac0c011c58becccd5d9ff15f358251b4437
This works since each test case has a unique working folder, so no conflicts can happen.

kraigher, 2016-07-22:
When I try to run parallel simulations using tests that execute in the same test bench myself I get:
```vhdl
# ASDB: Error: remove: "vunit_out\rivierapro\dataset.asdbw\current\global_info": The process cannot access the file because it is being used by another process.
```
I have fixed this by ensuring each simulation writes it own local dataset.asdb

kraigher, 2016-07-22:
I have pushed the fix to master in pull request #184

closed 2016-07-22 by kraigher