VUnit/vunit #167: Problems running run.py in Windows 7 with python 3

RomelTorres, 2016-04-27:
I am running the user guide vhdl example, and it does not work. I get an error while it tries to write the modelsim_ini: 

self._create_modelsim_ini()
  File "C:\Anaconda3\lib\site-packages\vunit\modelsim_interface.py", line 154, in _create_modelsim_ini
    write_file(self._modelsim_ini, read_file(join(self._prefix, "..", "modelsim.ini")))
  File "C:\Anaconda3\lib\site-packages\vunit\ostools.py", line 270, in write_file
    file_to_write.write(contents)
  File "C:\Anaconda3\lib\encodings\cp1252.py", line 19, in encode return codecs.charmap_encode(input,self.errors,encoding_table)[0] UnicodeEncodeError: 'charmap' codec can't encode characters in position 3273-3274: character maps to <undefined>

The workaround I found was to manually copy the modelsim.ini file into the vunit_out/modelsim directory.

kraigher, 2016-04-27:
Are you using Python 2.7 or 3.x?

RomelTorres, 2016-04-28:
Hi I am using Python 3.5

So far what I did was to change this =>  
_cfg.read(self._modelsim_ini)_ for 
_cfg.read_file(self._modelsim_ini, encoding = "utf-8") _
in lines number: 233 and 243 in the vunit/vunit/modelsim_interface.py. It is running so far, basically the problem is that the default encoding (sys.getdefaultencoding()) in windows is cp1252. I think in Python 3 you got to specify the encoding type.

DennisLipschinski, 2016-11-11:
I have the same problem. Also on the latest release, the potential fix, moves the error from _create_modelsim_ini to configparser.py

my suggestion is:
remove the fix and add a ", errors='replace'" to the open function in write_file

kraigher, 2016-11-15:
This should be fixed now.

closed 2016-12-03 by kraigher