VUnit/vunit #169: vunit.__doc__

vermaete, 2016-05-18:
Would it be better to have the description in the setup.py file in the docstring of 
[__init__.py](https://github.com/VUnit/vunit/blob/master/setup.py#L59)?

```
>>> import vunit
>>> print(vunit.__doc__)

Public VUnit interface
```

LarsAsplund, 2016-06-10:
@vermaete The long description in setup.py uses README.rst which exists to have an introduction generated on [PyPI](https://pypi.python.org/pypi/vunit_hdl). To avoid maintaining several text I'd like to reuse that in `__init__.py`

``` python
__doc__=open(join(dirname(dirname(__file__)), 'README.rst')).read()
```

This will look a bit weird due to the rst format but it's still readable. Acceptable?

```
>>> print(vunit.__doc__)
What is VUnit?
==============

VUnit is an open source unit testing framework for VHDL/SystemVerilog
released under the terms of Mozilla Public License, v. 2.0. It
features the functionality needed to realize continuous and automated
testing of your HDL code. VUnit doesn't replace but rather complements
traditional testing methodologies by supporting a "test early and
often" approach through automation.

**Read more on our** `Website <https://vunit.github.io>`__

License
=======

.. |copy|   unicode:: U+000A9 .. COPYRIGHT SIGN

VUnit
-----

VUnit except for OSVVM (see below) is released under the terms of
`Mozilla Public License, v. 2.0`_.

|copy| 2014-2016 Lars Asplund, lars.anders.asplund@gmail.com.

OSVVM
-----

OSVVM 2015.03 is `redistributed`_ with VUnit for your convenience. Minor
`modifications`_ have been made to enable GHDL support. Derivative work
is also located under `examples/vhdl/osvvm\_integration/src`_. These
files are licensed under the terms of `ARTISTIC License`_.

|copy| 2010 - 2015 by SynthWorks Design Inc. All rights reserved.

.. _Mozilla Public License, v. 2.0: http://mozilla.org/MPL/2.0/
.. _redistributed: https://github.com/VUnit/vunit/blob/master/vunit/vhdl/osvvm
.. _modifications: https://github.com/VUnit/vunit/commit/25fce1b3700e746c3fa23bd7157777dd4f20f0d6
.. _examples/vhdl/osvvm\_integration/src: https://github.com/VUnit/vunit/blob/master/examples/vhdl/osvvm_integration/src
.. _ARTISTIC License: http://www.perlfoundation.org/artistic_license_2_0
```

vermaete, 2016-06-10:
it's pefect!

Is there a way to have the version info that is in the setup.py also copied to the __init__.py?
I did not found another way to check the version of vunit than: pip list | grep vunit.
But this works only when vunit is installed with pip and does not allow me to put a python assert in the 'run.py' on the version of vunit.

LarsAsplund, 2016-06-10:
The only way to be sure about the version is if it was installed with PIP. If its a clone from GitHub you will have the latest and greatest. If it was installed with PIP it would be possible to do something like

```
>>> import pip
>>> installed_packages = {i.key:i.version for i in pip.get_installed_distributions()}
>>> print(installed_packages['vunit-hdl'])
0.52.0
```

but that will only work for those having pip installed. I think the earlier Python versions we support didn't included pip in the distribution so this is nothing that we can assume. Also, as you can see from the example above the version can be misleading when you normally use a Git clone. The latest version I installed with pip was 0.52.0. I have to think about this a bit more.

suoto, 2016-06-10:
@LarsAsplund @vermaete For version string management may be worth checking out [warner/python-versioneer](https://github.com/warner/python-versioneer).

It has some caveats, but you would be able to do

```
$ git clone https://github.com/VUnit/vunit
$ cd vunit
$ python -c 'import vunit; print vunit.__version__
v0.66.0
```

However, you **will not** be able to install via pip **and** use the version string within Python but pip would report the correct version. (versioneer shows `0+unknown` when it can't find the proper info to build the version string.)

```
$ pip list | grep vunit
vunit-hdl (0.66.0)
$ pip install git+https://github.com/VUnit/vunit
$ python -c 'import vunit; print vunit.__version__
0+unknown
```

LarsAsplund, 2016-06-10:
@vermaete @suoto The issue is well covered [here](https://packaging.python.org/en/latest/single_source_version/#single-sourcing-the-version). I'm going to read it later and then suggest something

LarsAsplund, 2016-06-12:
@vermaete I added the `__doc__` string as well as a `__version__` to the `__init__.py` file. You can also use the `doc()` and `version()` functions in `vunit.about` to get the same information. Acceptable?

joshrsmith, 2016-06-13:
@LarsAsplund I have read various warnings about querying version information at runtime as you have proposed here. I think the issue resolves around how the version might be reported differently when the package is installed vs develop vs running from source.

See http://stackoverflow.com/questions/17583443/what-is-the-correct-way-to-share-package-version-with-setup-py-and-the-package

More discussion here: http://stackoverflow.com/questions/2058802/how-can-i-get-the-version-defined-in-setup-py-setuptools-in-my-package

LarsAsplund, 2016-06-14:
@vermaete @joshrsmith @suoto Let's see if we can work out the user stories and verify that those are met.
* We want to get `__doc__` and `__version__` from the `vunit` package and to make that work when you have several versions installed/cloned it should be part of the `vunit` package you have imported and not something you get from `pip` or similar.
* For maintenance reasons I want single source. The exception is `README.rst` which has to exist to get the basic information presented in the Github [repo](https://github.com/VUnit/vunit). I added a [test case](https://github.com/VUnit/vunit/blob/master/vunit/test/lint/test_readme.py) which verifies that  `README.rst` == `__doc__` to avoid mistakes.
* Then we have the case of installing/cloning VUnit. It has to work regardless if you have an existing VUnit version or if it is the first time. Installing with `pip` without any existing VUnit on the path works for  me. The 

``` python
from vunit.about import version, doc
```
 statement in `setup.py` is handled so I assume that its directory is on the path used when running `pip`. It also works if I run `setup.py` directly but only as long as I'm running from that directory. That's a bit of a weak spot. Maybe I should read and call `exec` with the `about.py` code directly rather than importing it.
* Another issue is that if you're cloning VUnit `__version__` shouldn't return `0.66.0` as it is doing today. The repo has changed after that release. I think the best way of working, if we want to be strict, is to change the version directly after a new release has been uploaded to PyPI. So if 0.67.0 is uploaded the version should be immediately changed to 0.68.0-rc.1. Anyone checking versions automatically should be able to handle these formats. The rules about this are defined by http://semver.org/

vermaete, 2016-06-14:
I never thought that a version would give me so many deep Python readings -)
Anyhow: I'm only interested in the version of vunit installed with PIP.
I have to assure that a VHDL project is build with the tools and versions as described in one of the so many documents.  Tools itself, as vunit, has to be installed in a way it can be re-done by the client.  And explaining git/gitlab/... is far beyond the scope. 

So having Travis, having let it build the python package en let it fill in the version number?  And upload it?

Then I can write :assert vunit.__version__ == "0.66.4"

LarsAsplund, 2016-06-14:
@vermaete Travis let us have automatic sanity checks on people's pull requests among other things. However, when to create and upload a new release is manual work.

Anyway, what we have now should let you verify your installations.

closed 2016-06-15 by LarsAsplund