VUnit/vunit #170: Modelsim 10.5 "modelsim.ini" UnicodeEncodeError

bleviet, 2016-05-19:
Hi!
Since update from Modelsim 10.4b to Modelsim 10.5, I cannot start simulation after executing run.py with my design as well as VUnit example design. Please see the following console output:

> Traceback (most recent call last):
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\ui.py", line 628, in main
>     all_ok = self._main()
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\ui.py", line 659, in _main
>     simulator_if = self._create_simulator_if()
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\ui.py", line 727, in _create_simulator_if
>     return self._simulator_factory.create()
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\simulator_factory.py", line 160, in create
>     self._args)
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\modelsim_interface.py", line 87, in from_args
>     gui_mode="load" if args.gui else None)
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\modelsim_interface.py", line 122, in __init__
>     self._create_modelsim_ini()
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\modelsim_interface.py", line 154, in _create_modelsim_ini
>     write_file(self._modelsim_ini, read_file(join(self._prefix, "..", "modelsim.ini")))
>   File "C:\Python27\lib\site-packages\vunit_hdl-0.66.0-py2.7.egg\vunit\ostools.py", line 270, in write_file
>     file_to_write.write(contents)
> UnicodeEncodeError: 'ascii' codec can't encode characters in position 3195-3197: ordinal not in range(128)

Best regards,
Bach

RomelTorres, 2016-05-19:
I had an issue similar  (Issue #167) I solved it by specifying in the read function the encoding type. so far I haven't had any other problems, I thought it was Windows/Python related, but it could be that the default encoding of the .ini file changed?

LarsAsplund, 2016-05-19:
I don't see this when upgrading to the latest Modelsim Altera Edition but that has vcom/vsim 10.4d. What happens if you start fresh by removing the `vunit_out` directory?

bleviet, 2016-05-20:
Thank you for your answers!

With the **"potential fix"** commit from LarsAsplund it **works fine** again. Thank you very much!!!

Note for RomelTorres:
I tried the encoding="utf-8" option you mentioned but it doesn't help. I've checked the modelsim.ini encoding type in Notepad++ for both version and they're same (UTF8).

closed 2016-05-20 by bleviet