VUnit/vunit #171: name="testsuite" in xUNIT

vermaete, 2016-05-20:
I would like to run my testbench 3 times.  Once at RTL, postsynthesis and postlayout level.
It's of course the same testbench and the testbench is compiled into the 'tb_lib' library.  Only the library of the DUT is compiled again.
Those 3 runs are generating 3 xUnit xml files ['rtl.xml', 'postsynth.xml', 'postlayout.xml'].

```
# Extra argument parsing on top of VUnit
cli = VUnitCLI()
cli.parser.add_argument('--mode',
                        choices=['rtl',
                                 'postsynth',
                                 'postlayout',
                                 'postlayout_sdf'],
                        default='rtl',
                        help='level of simulation')
args = cli.parse_args()
args.output_path = "vunit_out_{}".format(args.mode)
```
```
x_lib = vu.add_library("{}_lib".format(PROJECT))
if args.mode == 'postsynth':
    x_lib.add_source_files("../implementation/{target:}/synthesis/{component:}.vhd".format(target=TARGET,
                                                                                               component=COMPONENT))
elif (args.mode == 'postlayout' or
      args.mode == 'postlayout_sdf'):
    x_lib.add_source_files("../rtl/{0}_vhd_pkg.vhd".format(COMPONENT))
    x_lib.add_source_files("../implementation/{target:}/designer/{component:}/{component:}_ba.vhd".format(target=TARGET,
                                                                                                              component=COMPONENT))
else:
    x_lib.add_source_files("../rtl/*.vhd")
```

Problem is that the content of the XML of those runs isn't reflecting the thee modes.  
```
<testsuite errors="0" failures="0" hostname="XYZ" name="testsuite" skipped="0" tests="1">
  <testcase classname="tb_lib.tb_unit" name="testcase_bfm" time="2.3">
```

And Jenkins is confused when displaying the test results with the xunit plugin.

A solution could be, but please guide me in this, is to to have the name="testsuite" configurable.  At the moment the string 'testsuite' is [hard code](https://github.com/VUnit/vunit/blob/master/vunit/test_report.py#L191).  
Another one would be to change the name of the library (tb_lib).  But this would make it harder to publish my test results.

LarsAsplund, 2016-05-21:
What you can do is to add a configuration to your testbench. Something like

``` python
tb_unit=tb_lib.entity("tb_unit")
tb_unit.add_config(name=args.mode)
```

and your XML should be like this

```
<testsuite errors="0" failures="0" hostname="XYZ" name="testsuite" skipped="0" tests="1">
  <testcase classname="tb_lib.tb_unit.<mode>" name="testcase_bfm" time="2.3">
```

vermaete, 2016-05-21:
Your solution is simple.  It's good and it's perfect! Tnx
[image]

closed 2016-05-21 by vermaete

LarsAsplund, 2016-05-21:
Great. I recommend that you have a look at the API documentation for `add_config` as well. There are more things you can do. For example running with different generics. You'll find an example of that [here](https://github.com/VUnit/vunit/tree/master/examples/vhdl/generate_tests)