VUnit/vunit #173: Get the testname?

vermaete, 2016-06-02:
I would like to pass a generic to the VHDL testbench that has the name of the test in it.
Can this be done? 

```
entity tb is
    generic (
        vectfile: string := "default.vec");  -- This should be overruled in run.py
end entity tb;
```
```
  p_stim : process is
  begin  -- process p_stim
    test_runner_setup(runner, runner_cfg);

    while test_suite loop

      wait until rising_edge(devrst_n);
      wait until rising_edge(clk);

      ---------------------------------------------------------------------------
      -- test case
      ---------------------------------------------------------------------------
      if run("testcase_bfm") then
        info("Running the BFM test cases");
        wait until bfm_finished = '1';

      elsif run("testcase_pattern") then
         -- another test....
      end if;
    end loop;
    test_runner_cleanup(runner);        -- Simulation ends here
  end process p_stim;
```

```
from vunit import VUnit

vu = VUnit.from_argv()
vu.set_generic("vectfile", "../bfm/{:s}.vec".format(THE_TING_I_DONT_NOW))  

tb_lib = vu.add_library("tb_lib")
tb_lib.add_source_files("../testbench/*.vhd")
vu.main()

```

kraigher, 2016-06-02:
Could be done like this:
```python
tb = lib.entity("tb_of_interest")
tb.set_generic("vectfile", "../bfm/default.vec")
for file_name in glob("../bfm/*.vec"):
    test_name = splitext(basename(file_name))[0]
    tb.test(file_name).set_generic("vectfile", file_name)
```

LarsAsplund, 2016-06-02:
Can also be done in VHDL

``` vhdl
entity tb_demo is
  generic (
    runner_cfg : string;
    path : string := "my/default/path");
end entity tb_demo;

architecture tb of tb_demo is
begin
  test_runner : process
  begin
    test_runner_setup(runner, runner_cfg);

    while test_suite loop
      report "File to load: " & join(path, active_test_case & ".vec");
      if run("test_1") then
      elsif run("test_2") then
      end if;
    end loop;

    test_runner_cleanup(runner);
  end process test_runner;
end;
```

Results in

``` console
...
# ** Note: File to load: my/default/path/test_1.vec
...
# ** Note: File to load: my/default/path/test_2.vec
...
```

LarsAsplund, 2016-06-09:
@vermaete Does this answer your question such that the issue can be closed?

vermaete, 2016-06-10:
Sorry, I didn't found time yet to try.  I was planded to take the Python solution.  But I'm confident the solution will work.  You may close this ticket.

closed 2016-06-10 by vermaete