VUnit/vunit #175: Adding colors to logging output

suoto, 2016-06-21:
I've been testing colorized VUnit logging so that the output log shows different colors depending on the severity level. The goal is to make easier to spot interesting stuff on large logs.

Below is a capture of how the logs look like:

[image]

Since the code is roughly done, I would like to ask feedback on some issues before diving into fixing the unit tests that are currently failing.

* The coloring scheme looks good enough for me, but you might want to do some fine tuning. I'm not sure how to implement a way for the user to change colors
* User should be able to enable/disable colored logs via run script. Is it OK to user the `--no-color` to control this as well?
* On ModelSim (and probably other simulators), the raw control chars are shown at the transcript window as shown below. To avoid this, using `--gui` must imply `--no-color`. Does this sound OK?
  [image]

cmarqu, 2016-06-21:
(User input here.) 
I quite like your screenshot. It looks like it could be made to also work nicely in a terminal with bright background (like I am using).
However, color rendition is rather different between terminals (tried xterm, kterm, gnome-terminal at some point), and depends on advertised capabilities as well (i.e. the same basic colors look different in my gnome-terminal with and without "TERM=xterm-256color" set), so I can imagine the usefulness of full configurability to make it look nice everywhere.
(As for implementing that... Python config options forwarded as string VHDL generics/Verilog parameters maybe?)

suoto, 2016-06-22:
@cmarqu Thanks for the input. :)
Indeed, different terminal capabilities should be handled and tested. I'm thinking on how to communicate that to VHDL; I guess Python could write/modify something like an 'environment info' package with that. I'll give it a try to check if that could actually work.

LarsAsplund, 2016-06-22:
Wouldn't it be better to capture the output from the simulator, look for warning/error/failure keywords and highlight those lines? Then we would avoid the lack of such functionality in the simulator transcript windows and we would reuse the color printing we already have for pass/fail status/summary. I'm hoping to get some input from @kraigher on the implementation details  on this since he wrote the relevant functionality.

suoto, 2016-06-23:
@LarsAsplund, adding colors outside the HDL seems rather interesting. Only that instead of capturing the simulator output directly, I think it's more reliable to use a CSV-like output.

The way I think to accomplish that is changing the HDL log messages to log the message in CSV format. On the Python side, the CSV log record is converted to the log string with or without colors depending on the terminal. Writing to files would always be without colors. This seems feasible, but I'm not accounting for other side effects.

@LarsAsplund and @kraigher, would you please check if my understanding is correct?

LarsAsplund, 2016-06-23:
@suoto Have a look at the `consume_output` method in the `Process` class in [ostools.py](https://github.com/VUnit/vunit/blob/master/vunit/ostools.py). A color coding callback function would probably do what you're looking for.

suoto, 2016-06-24:
@LarsAsplund I was actually testing with the `TeeToFile` class inside [test_runner.py](https://github.com/VUnit/vunit/blob/master/vunit/test_runner.py#L204) because it knows the streams it is writing to and whether it supports special chars.

To avoid confusion, I added `verbose_colors` to the `log_format_t` to change the log format a little so the Python side knows when it should to something.

kraigher, 2016-06-24:
@suoto The Python side of VUnit is not integrated with the VHDL logging mechanism in any way today. The Python side just blindly captures what is printed to stdout/stderr during simulation. If you in VHDL choose to write a log-file to a csv file using the log package it just becomes a csv file on disk that the Python part does not know or care about.

Adding a general output filter somewhere in the simulator independent parts of the Python code that detected lines with error or warning words in them for coloring would be the easiest solution. It would be unintelligent though since it cannot now the intention of the line being printed such as if it really was a warning or just contained the word warning.

NOTE: Your initial VHDL-based solution will not work on Windows without using Cygwin or similar since the Windows console does not use in-stream escape characters to change color but rather a Win32 API.

suoto, 2016-06-24:
@kraigher Thanks for your input.

To make simulator independent and avoid coloring wrong strings, I was thinking of doing the following:

1. On the testbench, user configures the logger to use colors if possible

    ```python
    logger_init(display_format => verbose_colors);
    ```

1. The `log_formatting` package use pretty much the same `verbose_csv` format to print to stdout, only that it prepends a string so the Python side knows it needs to format from CSV to "real" log and colorize if possible. In the example below I used `c` as the first value of the CSV. An example output would look like

    ```
    # c,101,100 ps,verbose,,,,Verbose messages should not stand out
    # c,103,100 ps,debug,,,,Debug messages should not stand out either
    # c,105,100 ps,info,,,,Info messages use no highlighting to look like the rest of the log
    # c,107,100 ps,warning,,,,Warnings on yellow
    # c,109,100 ps,error,,,,Errors on red
    # c,111,100 ps,failure,,,,Failures are easy to spot
    ```

  I know that ModelSim inserts the `#` character at the end of each line and GHDL don't, so the format above is just an example.

1. The `TeeToFile` class receives output from the simulator. If the string has a CSV record whose first character is `c`, format the record. If the current stream support colors, formatting the record includes colors as well.

   If the string doesn't have the `verbose_colors` identification, no changes are done

I did a proof of concept of this on my fork and it seems to be working. Could you please check if the above description has any holes or if there are better alternatives?

kraigher, 2016-06-25:
@suoto The general idea would work. Here are some issues I see: 

1. It would not color the built-in errors and warnings generated by the simulator itself. Maybe that was out of your scope for this. 

2. When running in `--gui` mode the user might not want to see the unprocessed CSV format in the simulator transcript window so this feature has to be automatically switched off then.

3. In my opinion it would not be good software design to make coloring and reformating CSV a concern of the `TeeToFile` class. I would prefer an isolated output postprocessing wrapper object was added around the stdout object.

suoto, 2016-06-25:
@kraigher 

> (1) It would not color the built-in errors and warnings generated by the simulator itself. Maybe that was out of your scope for this.

Yes, that was out of the scope indeed. It could be done but using a different approach I guess.

> (2) When running in `--gui` mode the user might not want to see the unprocessed CSV format in the simulator transcript window so this feature has to be automatically switched off then.

I'm considering that if the user ran with `--gui`, the behaviour of `verbose_colors`
should be equal to `verbose`.

> (3) In my opinion it would not be good software design to make coloring and reformating CSV a concern of the `TeeToFile` class. I would prefer an isolated output postprocessing wrapper object was added around the stdout object.

Agreed, that's why I referred to it as a proof of concept. :)

LarsAsplund, 2016-06-26:
The purpose of highlighting is to quickly spot all probIems in a (long) log and to do that we must address  all sources of problems. We should highlight compile time as well as run time problems, regardless of the simulator and the log/check solutions being used. I also think this is a very good feature which should be activated by default just like the coloring we have for pass/fail today.

This is best done by having a Python-based solution looking for specific words/phrases so I  don't think we should have special formatting that needs to be activated in the testbench. I did a very quick test and just looking for error/warning/failure/fatal followed by a non-alphanumeric character takes you a long way. It will do good detecting compile warnings and errors when using the simulators we support and it works well for VUnit check/logs, VHDL assert, and OSVVM/BVUL/UVVM solutions. It spots problems found by checks/asserts added to the testbench as well as problems detected by the simulator, for example division by zero. A few exceptions I found in this quick test
* When running VUnit under VHDL-93 the assert stopping the a successful simulation is highlighted. Adding some report statement to that assert like `"Exiting VUnit simulation"` would make it easy to spot that exception.
* Compile errors with GHDL results in `Failed to compile ...`. Whether such phrases should be included by default or be simulator specific is up for discussion

Regardless of the intelligence being built-in there will be false detections but I don't really see that as a major problem. Usually when you look at a log it is because you have a test that failed and if that log has two rather than one highlighted lines where one is a false detection you will still able to quickly spot the real issue. I rather have this situation than a solution which fails to highlight the problem line due to a too specific detection pattern 

For those that still want to improve on the default behavior there should be a way to configure the detection pattern. I don't have experience with very many tools supporting generic problem highlighting but the plugin for Jenkins I know of uses this approach so it seems to be a generally accepted solution.

suoto, 2016-07-01:
I would like to share an update on this. The functional aspects seem OK. I'm using this version daily at work and have manually tested some of the added features. I have adapted the VHDL logging test and will add tests for the Python side before creating a PR.

(Please not that currently this only includes adding colors to the VHDL logging system, not the highlight of built in errors and warnings from simulators.)

In the meantime, I would like to share a summary of the changes so you can point issues I may have missed.

* `vunit/ui.py`:
  * `VUnit` class passes the `--gui` argument and its own printer to the `TestRunner` when running tests
* `vunit/test_runner.py`:
  * Added `CsvLogFormatter` class to format and colorize log records
  * Added 2 arguments to the `TestRunner` constructor:
    * `printer`: passed on to CsvLogFormatter objects
    * `on_gui`: used when running the test suites so the `runner_cfg` generic includes a `running on gui : True/False` field. This disables printing special characters on the transcript window
* `vunit/color_printer.py`:
  * Defines the colors used when formatting the log record
  * Log colors can be changed via `VUNIT_COLORS` environment variable
* `vunit/test_report.py`:
  * Removing control chars when generating XML at `TestResult.to_xml` method
* `vunit/vhdl/run/src/run.vhd`:
  * Uses the `running on gui` parameter to fallback to `verbose_no_colors` when running on GUI
* `vunit/vhdl/logging/src/log_formatting.vhd`:
  * The `verbose_format` procedure formats the log record into a CSV format that will be formatted by the `CsvLogFormatter` class on the Python side
* `vunit/vhdl/logging/src/log_types.vhd`:
  * Added log format `verbose_no_colors` to revert to the previous mode
* `vunit/vhdl/logging/test/tb_logging.vhd`:
  * Added tests for the CSV log format used for coloring

Please let me know if you have any suggestions.

kraigher, 2016-07-05:
@suoto Looking at your pull request, even though it is well implemented, I must say I am not convinced the benefits of this feature outweighs the added complexity and extra cross-cutting concerns between the Python and VHDL side of things. As a maintainer of VUnit this is a concern for me and I am leaning towards saying the cost of the solution outweighs the value of it. @LarsAsplund any opinion from your side.

joshrsmith, 2016-07-06:
how does this solution work cross-platform? e.g. How will logging work on Windows cmd.exe vs Cygwin vs Linux vs Mac?

suoto, 2016-07-06:
@kraigher and @Lars et al., Looking at the current implementation and comparing to the idea of implementation I had when I opened this issue, I must confess I thought it was easier to get it done. If it worth the effort, I'm obviously open to fix issues that may arise.

@joshrsmith Cross-platform should be OK because it uses the `color_printer.py` developed by kraigher to add colors to the pass/skip/failed lines on the report. Only that I have tested just now on a Win 7 VM and there are some issues (format is OK but no colors; I think I know why). I haven't tested on Cygwin or Mac. On Linux so far it's working fine.

suoto, 2016-07-07:
@joshrsmith The Windows issue has been fixed on 8f6e7fe5237ca50904e6e617cab88fa65fab07f8

LarsAsplund, 2016-07-18:
@kraigher I think this type of coloring has value which makes it worth some effort. I've kept the quick fix I did and it did draw my attention to issues which I almost forgot about, for example the warnings of empty package bodies that GHDL issues when compiling VUnit. So, it's the attention catcher it's supposed to be.

I agree that we should keep the Python side and the VHDL side as separated as possible. Using a pure and more generic Python solution will reduce the dependencies as well as increase the value. The major problem I see is if we end up with many false positives. @suoto et al, what are the phrases we want to highlight and what are the phrases likely to become false positives?

I'm ok with aiming for a partial solution to start with but the first use case to handle should be 

```ERROR: Something is wrong```

since errors are what we primarily log and this is the default format. Old testbenches should get this feature by default without changing the code to have it enabled. Just looking for `ERROR` followed by a space or a colon will also highlight outputs on the verbose format. @suoto, your solution takes it even further and uses different coloring for different fields. This can be done as well but it will not work with other logging solutions and is the added value of having different coloring for different fields that high? Isn't the fields of

```1 ns: WARNING (my_file.vhd:17): Some message```

rather distinct without specific coloring?

suoto, 2016-07-19:
@LarsAsplund  Phrases I know that could become false positive are ModelSim's report formats. Because the user can change it via `modelsim.ini` file, this could be an issue. I'm assuming that if the user _is able_ to change, then someone _will eventually_ change, so this should at least have some specific testing.

On the false negatives side, log levels that were renamed won't be highlighted.

But I guess if a Python only approach makes things much simpler then it is worth the trade offs.

When you say 'work with other logging solutions' you mean OSVVM `AlertLogPkg`, right? The different colors for each field I considered it was a nice to have feature, but I'd choose VUnit behaving cohesively across multiple logging solutions. Please note that I haven't used the `AlertLogPkg` so is hard for me to say anything.

LarsAsplund, 2016-07-19:
@suoto If we were to look for words like failure, fatal, warning, and error followed by a non-alphanumeric character and the assertion message includes any of these words without being a message of that severity we would have problems. The ModelSim message format could be changed to include such words even though the message being printed isn't of that level of severity but I can't see that this would be a major problem. I would expect notes like "error count: 0" or notes from an instance named `error_handler` to be a bigger problem. If that happens people can accept the false positive (the test case will still pass), change the sting/naming, use `--no-color`, or we could add configuration to control highlighting behaviour.

New or renamed log levels would be ignored but in general we can't make any assumptions on how such a level should be handled. For example, if someone renames `warning_low2` to `issue` we cannot know if that should be highlighted or not and what color to use. This has to be controlled by configuration.

The other logging solutions I'm referring to are primarily VHDL report/assert but also OSVVM and the Bitvis Utility Library (BVUL). We try to avoid lock-in situations where you have to use the VUnit solutions in order to have access to some features. Considering that the BVUL/OSVVM promoted way of doing testbenches involves logging more or less everything that happens and put that on stdout I'm imagining that they would benefit from coloring even more to easily spot the more interesting parts of long logs. They do not have the possibility to log the details to file (normally just interesting when debugging) while just showing more severe messages and progress information on stdout. Using the keyword I mentioned previously seems to work well for these libraries as well.

Btw, have you looked for already existing Python packages for we are trying to do?

kraigher, 2016-07-19:
After considering this for some time I think the cleanest solution is to actually emit the ANSI color codes from VHDL directly just as your original proposal making the functionality completely local to the logging package without any complicated coupling between VHDL and Python. Then it is easy to add colorized printing of other stuff as well using the same mechanism. Windows users can initialize the [colorama](https://pypi.python.org/pypi/colorama) package in their run.py to get the ANSI colors to work.

I do not like the tight coupling in the current pull request that there is a specific format from a VHDL logging package that must be parsed the generic TestRunner class. It is also not very general and would have to be extended if other things should be printed in color in the future. I think this functionality is not worth the extra maintenance cost of introducing this coupling. As a maintainer of VUnit with little time to spare this is a concern for me.

suoto, 2016-07-19:
That solves the coupling but emitting colors from VHDL directly results on a dirty transcript output as shown on the original comment unless (1) the test bench code knows when to skip emitting colors or (2) we could intercept strings before they go to transcript.

With regard to (1) it could be done by the VHDL code reading from a file if it should add colors or not. This file would be placed on a known location and have a format that's easy to parse. This would allow using `colorama`, emit control chars from VHDL and keep the transcript clean. Does this sounds good? 

---

With regard to coloring the compilation messages, it's a matter of just parsing the compilation output.

LarsAsplund, 2016-07-20:
@kraigher Do you include coloring of other simulator output, for example assert statements and compile errors, in "other stuff"?

kraigher, 2016-07-21:
No only coloring of ANSI codes. I do not think it is worth the maintainance burden to have a colorized output parsing. To many simulators, to many false positives, hard to test... No simulator manufacturer has this feature. We should not make efforts to work around it. It should be added in the simulator I n that case.

suoto, 2016-07-21:
A note for highlighting compile errors: I haven't tested PR #183, but if VUnit should not emit color codes when compiling (or doing whatever action) directly from the transcript window.

Adding colors to the compilation errors could be based on the successful execution of the compilation command (return code 0), i.e., if not 0, highlight everything. This does not mix Python with VHDL.

Wrt logging colors at runtime, I agree with @kraigher about being hard to test, especially acceptance testing under Windows. Colorama itself mocks Windows side to test on Travis, so they have no "real" acceptance CI on Windows platforms.

LarsAsplund, 2016-07-21:
@kraigher, @suoto I keep coming back to what I think is the main user story to highlight "issues" for faster browsing of all VUnit output. If only 50% or so of those issues are found the solution for that story is severely flawed. A solution that will will find most issues and a few extra is also flawed but still much better. I think a rather simple set of keywords and maybe a few more complex tailored  for VUnit logging will take us far and a simple set for the simulator output will also be less vulnerable to changes between simulator releases.

Even if we start small just supporting VUnit logs the VHDL approach will run into problems when we decide to extend the support. Users and developers would have to deal with some coloring in the VHDL code and simulator coloring elsewhere.

I agree that generic coloring support doesn't fit well into the Python test runner either so why not have a separate Python module that wraps Colorama in the same way it wraps stdout (Colorama is actually not needed in Windows 10 which supports color codes natively)? Such a module will be imported/activated in the run script and it will be decoupled from the rest of VUnit

LarsAsplund, 2016-07-28:
Here is a regexp that finds the verbose and level formats of VUnit and the individual fields of these.

``` python
custom_levels = [] # Allows configuration of regexp
standard_levels = ['FAILURE', 'ERROR', 'WARNING', 'INFO', 'DEBUG', 'VERBOSE']
time_units = ['fs' ,'ps', 'ns', 'us' ,'ms', 'sec', 'min', 'hr']
vunit_log_pattern = re.compile(
    # A time specification is available in the verbose format but not in the level format
    '(?P<time_spec>' +
        '(?P<time>\d+ (%s))' % '|'.join(time_units) +
        ': ' +
    ')?' +
    '(?P<level>%s)' % '|'.join(standard_levels + custom_levels) + # Standard or custom level
    # A location specification is optional in the verbose format and has a log source ID and/or a file location
    '(?P<location_spec> in ' +
        '(?P<src>[^:(]+)?' +
        r'(\((?P<file_name>[^:]+):(?P<line_num>\d+)\))?' +
    ')?' +
    ': ' +
    '(?P<msg>.*)') # The message is the rest of the line
```

It will also find failing VHDL asserts from Aldec's simulators since their format is identical to the VUnit level format.

Adding a regexp like this will cover most other interesting outputs

``` python
general_pattern = re.compile(r'(?P<level>fatal|failure|error|warning)[^a-zA-Z]',re.IGNORECASE)
```

but there is a need for simulator specific patterns to do really good. For example

``` python
ghdl_compile_error_pattern = re.compile(r'Failed to compile');
```

cmarqu, 2016-07-29:
For the record, for Cadence Incisive, the regexpes are: 
Warning: ``"\\*W,[A-Z0-9]*( \(.*\))?:"``, Error: ``"\\*E,[A-Z0-9]*( \(.*\))?:"``, Failure: ``"\\*F,[A-Z0-9]*( \(.*\))?:"``
In addition, one wants to highlight also "timing violation" and assertion errors which follow another syntax, but I am not entirely sure of their exact spelling right now.

suoto, 2016-08-04:
I have implemented a way to let the simulator specific classes to add colors to the compilation output (the simulator interface parses and the base class does the actual coloring).

As usual, ModelSim and GHDL interfaces already add colors, but I don't have Active-HDL and Riviera PRO regexes to test. I believe theses regexes could be find tuned over time, so it's not a big deal right now.

I'll test a similar approach to color the simulation output as well.

---

Please forgive me for taking all this time to get back to you. I didn't wanted to split focus across different projects so I took the time to finish other things.

kraigher, 2016-08-05:
I have not seen your solution but I imagine problems with having the simulator classes emit color. First of all on Windows coloring requires system state changes through win api. Since the simulator output in both the non verbose and multi threaded cases is captured by the test runner class to be printed later it will not work.

LarsAsplund, 2016-08-05:
@suoto I've tested your latest implementation and errors and warnings from the compilation look good. Here are some examples from Rivera-Pro/Active-HDL to work on

```
COMP96 WARNING COMP96_0119: "The range is 'null range'." "d:\Programming\github\suoto_vunit\vunit\vhdl\osvvm\RandomPkg.vhd" 104 38
COMP96 ERROR COMP96_0015: "';' expected." "d:\Programming\github\suoto_vunit\examples\vhdl\com\vunit_out\preprocessed\tb_shuffler_lib\tb_card_shuffler.vhd" 35 3
# ELAB2: Fatal Error: ELAB2_0016 tb_card_shuffler.vhd (27): No value assigned to generic/parameter "foo".
```

I have a few comments from the user point of view (I'm going to leave the implementation discussion for a while)
* It seems like a failing check isn't colored unless I run with the `-v` option (at least not in ModelSim). Instead I see the message with the `<VUNIT_CSV_LOG>` tag.
* There is a need for a failure/fatal error type for messages like the last of the Riviera-PRO examples. Also exists in ModelSim (see below), not sure about GHDL. I think fatal and failure should have the same coloring.

`# ** Fatal: (vsim-3350) Generic "foo" has not been given a value.`
* The default output format (level) isn't highlighted. From GHDL

`ERROR: Cards loaded and received differ`
* Plain VHDL asserts are not colored. I think these are important since people run VUnit on legacy code and there are also reasons for using them in new code. For example, Vivado can detect some assert violations during synthesis so there are reasons to use these over VUnit checks in parts of the code.
```

majoh, 2017-03-01:
Colors to logging output and compile out would be great!

kraigher, 2017-07-19:
I have decided that this is best solved using ANSI escape codes for color in VHDL. I will investigate integrating VUnit with colorama python package which enables support for ANSI escape codes on Windows. This way we avoid any special handling of our VHDL log package messages which I would really like to avoid. @suoto I know you did some VHDL code regarding adding ANSI color codes to the VHDL log package. Maybe you could dig that up again? When can then also add a configuration to the runner_cfg to disable ANSI color codes when running in a simulator GUI which will not handle the color codes.

kraigher, 2017-07-20:
I added an `ansi_pkg` on the `color_logging` branch and did some initial testing on git bash on windows. 10  It colorizes the output just fine.

kraigher, 2017-08-05:
There is now complete color logging support for both Windows and Linux on a branch. It is general purpose so not only log messages gets their color handled correctly but anything written to stdout.

kraigher, 2018-02-12:
Color logging is now part of VUnit 3.0

closed 2018-02-12 by kraigher