VUnit/vunit #179: New library "configuration" method

imd1, 2016-07-08:
There's currently a "entity" method in the library class, which I use to get a test handle

tb_vunit = lib.entity("tb_vunit")
test = tb_vunit.test("Test")

I have examples of a single VHDL test entity with 2 VHDL configurations defining 2 different tests, and so I'd to use a library "configuration" method to access each configuration and hence the test handle for each configuration.

Is this possible in a future release?

kraigher, 2016-07-09:
First of all configurations would need to be able to be considered as runnable test bench top levels which they are not today. After that it would make sense to add a Python method you describe.

joshrsmith, 2016-07-15:
I would also be interested in this functionality -- to have a configuration be a runnable top level for VUnit.

joshrsmith, 2016-07-18:
I have been trying to figure out if there is a work-around for this issue.

I thought you might be able to make a wrapper testbench and instantiate a desired "top level" testbench configuration in the architecture of the wrapper. But the problem then is that I do not know how to get the _runner_cfg_ from the testbench wrapper generics to push down through the instantiated configuration.

MortenZdk, 2017-04-04:
Hi Josh, 
Please find an example attached that I made while trying to do a work-around.  
Best regards Morten

[VHDL - config_tryout.zip](https://github.com/VUnit/vunit/files/893158/VHDL.-.config_tryout.zip)

kraigher, 2017-04-04:
The work around for this would be to create a wrapper test bench and use the scan_tests_from_file public method.

LarsAsplund, 2023-08-27:
Implemented by #951

closed 2023-08-27 by LarsAsplund