VUnit/vunit #180: ghdl parallel execution runs slowly

nathanaelhuffman, 2016-07-11:
I'm just playing around with GHDL and vunit and I noticed some un-expected (to me at least) behavior.  When I run vunit with the -p3 option, the tests run significantly slower.  Is this normal?
[image]

kraigher, 2016-10-06:
Still a problem?

nathanaelhuffman, 2016-10-08:
I believe so, it's unclear to me if this a GHDL issue or vunit, but given the GHDL without parallelization ran faster I figured vunit would be the place to report.  I did not find anything that would make this go any shorter but after reporting didn't spend much time working on it since there hadn't been much of a response. It's also possible I'm doing something wrong but am not sure where to start looking, ideas welcome.

kraigher, 2016-10-08:
I cannot reproducera it. Could you run with log level debug?

nathanaelhuffman, 2016-10-11:
I played around with this a little more today. This is windows 7 and python 2.7 if that matters much.  I used the latest master ghdl that I freshly built today and had the same issues.
It is launching the requested number of ghdl instances and from the debug log I can see that it is correctly feeding the instances with the correct subset of tests.

I didn't see much of interest in the log-level debug other than seeing the two process ids spawn as expected.  If you want to see it I can scrub it and provide it but it basically shows the number of pid's spawning in parallel and then when those complete it moves on to the next set.

If I can get some time later this week I might try this on a linux box with more cores as an additional datapoint, maybe my 4 core windows machine is not the best test platform.  I tried -p 1, -p 2, and -p 3.  -p 1 is a couple of seconds slower than the vanilla run with no parallelization, maybe not statisitcally significant. -p 2 and -p 3 both take longer.

LarsAsplund, 2016-10-11:
I run with a quad core Windows 10 machine and don't see this effect with Python 2.7. Do you see the same effect when running the example code included in the VUnit distribution? What version of GHDL are you using?

nathanaelhuffman, 2016-10-16:
I've tried a number of versions in the .34dev series including a fresh build from master of GHDL that I did myself last week.  It's odd, there's not much going on in the simulation.  I'll have to play with it to see if I can find a non-proprietary set of code that reproduces to share, but I'm rather busy at the moment so I'm not sure when I'll find time.  GHDL isn't my primary simulator so this was more of a curiosity and I explore vunit anyway.

LarsAsplund, 2017-02-22:
@nathanaelhuffman Any news?

nathanaelhuffman, 2017-02-23:
I'll see if I can get to this on the weekend this week so we can know more or close this.  Since I don't use GHDL as my regular sim env I'll see if I can find an existing sim that will run rather than the toy sim I was using when I reported this.

nathanaelhuffman, 2017-03-11:
Let's just close this. I haven't had any time to continue investigating this and I don't seen any coming in the near future.

closed 2017-04-17 by LarsAsplund