VUnit/vunit #181: Incremental compile when running in --gui mode

KevinKes, 2016-07-16:
While developing vhdl I use the following approach:

- develop functionality
- develop the test case with VUnit
- run test cases with VUnit python framework
- one (or more) test case fails:
 - run specific test case with --gui flag
 - add some waveforms and do vunit_run to debug the problem
 - fix possible bugs in the code
 - close modelsim
 - re run specific test case with --gui flag so file(s) get recompiled and I can add waveforms again

Restarting the simulator is quite cumbersome. It would be a nice feature to have incremental compile in the simulator gui somehow. What do you think? Or can you recommend me using an other approach?

LarsAsplund, 2016-07-16:
The way I do it is to keep ModelSim open and then  run the VUnit script with --compile from another command window. Then I restart the simulation from the GUI.

KevinKes, 2016-07-17:
That indeed does the trick. But I don't like switching windows (yes I'm a bit lazy .. ) so I was thinking about a few options:

1. Write out a Tcl script for the specific simulator with a vunit_compile procedure which can be executed in the simulator transcript window just like vunit_run.
 **a**. Just write out the full compile order as simulator compiler commands
 **b**. Just a wrapper to run the VUnit run script with --compile

2. Running a background thread which recompiles whenever a file changes.

Option 1b seems low hanging fruit to me, if I can find some time in this or the next week I can help to implement it if you like it.

joshrsmith, 2016-07-17:
I don't have much to add, but I too have landed on using a flow similar to what @LarsAsplund suggests. Though it would be nice to be able to invoke recompilation without leaving your simulator.

KevinKes, 2016-07-18:
I've implemented a 'vunit_restart' function, available to use in the ModelSim transcript.
You can already review my commit https://github.com/KevinKes/vunit/commit/3dda964ffeffe33de8261725861814354c3650bc

kraigher, 2016-07-18:
@KevinKes does it catch the exit code and stop before doing `vunit_run` if the compilation failed?

KevinKes, 2016-07-18:
@kraigher no it does not do that (yet). I used it today at work a few times and it was quite convenient to have a restart function which does the compile, restart and run.

It indeed sounds like a good idea to catch possible compilation errors.

KevinKes, 2016-07-19:
Just checked the current implementation; it stops when a compile error occurs and does not restart the simulation.

kraigher, 2016-07-19:
Ok that is good.

joshrsmith, 2016-07-23:
I don't see a reason why we can't add this feature to Riviera. I will look into it, if someone else doesn't get there first.

joshrsmith, 2016-07-28:
Modeling after modelsim approach on this, it appears that there are pathing issues in Riviera. For example, when invoking run.py from riviera, it goes from its current working directory... Which creates a structure like this: vunit_out/rivierapro/vunit_out/rivierapro

Does Modelsim not behave in the same manner?

KevinKes, 2016-07-28:
The path I get with modelsim = `vunit_out/modelsim`.

joshrsmith, 2016-08-18:
What is the working directory that modelsim uses when simulating?

When Riviera runs, it's working directory is `vunit_out/rivierapro`. This causes a problem if re-using the command that started simulation (e.g. run.py) because it is not in the same directory. If you add the proper pathing for it, then the default output is generated in the wrong location.

kraigher, 2016-08-18:
Some central part of VUnit should provide all simulatorinterfaces with the re-compile command. The output path and the starting working directory can be taken Into account. The output path can be set with a command line argument. Care must be taken to run the Python script with the rigth interpreter using sys.executable.

kraigher, 2016-12-08:
This is fixed for RivieraPro and ModelSim on master now.

closed 2016-12-08 by kraigher