VUnit/vunit #182: Flow with Vivado IP Integrator / External libraries

joshrsmith, 2016-07-19:
# Background
Recently I have been working with VUnit and Vivado IP Integrator (IPI) and running into library management difficulty. I have had success with using VUnit to map external libraries for Vivado IP and then simulating the IP - for example, simulating a single piece of IP (compiled into an external library) instantiated in a testbench (compiled into a VUnit managed library).

The primary difficulty is when plain HDL sources (that I would like to be managed by VUnit controlled library) are mixed into the Vivado IP Integrator block diagram.

In this configuration, IPI automatically creates a wrapper around the HDL file per-instance, and in the wrapper the HDL is instantiated by component instantiation with a default binding (no libraries, configurations, or entities are explicitly referenced). From my understanding of VHDL, the default binding implies it will search in the library that the component is compiled in, for an entity of the same name and port map as the component.

I bring this up because when initially planning on working with VUnit and Vivado IPI, I was hoping that I would be able to compile Vivado block diagram IP (e.g. all the generic Xilinx IP + project specific IP configurations) into a set of externally managed libraries, and then use VUnit to compile HDL sources into VUnit managed libraries. The typical flow would then be:

- Structure IPI block diagram in Vivado
- Run a (long running) process to generate IPI output products and compile Vivado IP for the block diagram
- Run VUnit simulations on portions of the block diagram, incrementally compiling changes to custom HDL files as development/debug continues.

My assumption has been I will reconfigure IP/block diagram infrequently, and will iterate on custom HDL designs frequently, so optimizing the compile/simulate loop for custom HDL is important.

Again, working with a single piece of IP at a time is straightforward since it can simply be instantiated by itself in a VUnit managed testbench. Working with a group of IP (in a block diagram hierarchy) or IP and HDL code is more difficult since portions of the block diagram itself must be instantiated - and have likely already been compiled together into various libraries as part of the Vivado IP compilation step.

I did investigate using VUnit to manage _all_ sources, but this is not feasible since some IP are encrypted and Vivado must be queried for proper compilation order and sources compiled separately (this is done in one of the examples in VUnit).

# Suggestions

- I believe this issue could be addressed, at least partially, by #179 - If VUnit supported a configuration as a top level simulation, then an explicit configuration could be used to bind the block diagram components for custom HDL entities explicitly to implementations in a VUnit managed library instead of the default binding that would occur to an entity in externally managed Vivado IP library. In this case, at least initially, the custom HDL source would be compiled identically into two different places: Vivado IP (external) library, and VUnit managed library for simulation only. After changes were made, the incremental compilation of VUnit managed library would get updated, and the version in the external library would not get updated or used.
- Allow VUnit to adopt/manage libraries that have already been created. I can imagine a lot of reasons why an approach like this could cause problems (e.g. with understanding dependencies). However, it seems like if a unit has already been successfully compiled into a library, there is no reason why VUnit could not take ownership of that library and simply recompile the file back into the same library (without wiping out the other contents).

If anyone has similar experiences or approaches that have worked for them, please let me know.

kraigher, 2016-07-19:
I have worked with VUnit and Vivado for 2-3 years. Just as you do I typically have two steps where the first step generates and compiles all the behavioral simulation files of Vivado IP:s in the project like the Vivado example that ships with VUnit. The second step is then `run.py` where those libraries are added as external libraries. The first step is more time consuming without any support for incremental operations but is typically not run very frequently just when modifying or adding a new IP.

I have never created my own IP in Vivado IP-integrator containing my own custom HDL since I typically try to keep such tools at an arms length distance and avoid schematic design. Thus I have not run into the problem you describe. 

Your second suggestion should already be possible. The current implementation will actually work the way you describe if you first do `add_external_library` and then add your own custom files to it **again**.

joshrsmith, 2016-07-19:
@kraigher Thanks for the input. I thought I had tried to add files to an externally added library before, and ended up getting an error. I figured that it was unsupported altogether, so I didn't pursue any further. I will take another look and report back if I find something fishy.

As a side note, I agree with you on the advantages of keeping 3rd party IP and custom HDL designs at arms-length, but I haven't yet landed on the sweet spot of how to best achieve this without a lot of I/O at the top level of the block diagram.

kraigher, 2016-07-21:
It is not really something that we have thought about and explicitly support but reviewing the current implementation it looks like it will work the way you want it to.

joshrsmith, 2016-08-18:
Closing this issue -- Reasonable work-around available.

closed 2016-08-18 by joshrsmith