VUnit/vunit #185: GHDL : simulation options in wrong place on command line.

Brian-Drummond, 2016-07-22:
Using "set_sim_option" with ghdl's --elab-run command, the syntax expected by ghdl is that the elaboration options (e.g. "--std=08") are expected before the entity name, while the simulation runtime options (e.g. "--assert-level=none") must appear after it. 

However Vunit doesn't currently distinguish between elab and sim options, placing them both before the entity name on the command line.
Thus the following simulation options (elaboration options) work correctly, enabling code coverage via gcov (assuming ghdl built with the gcc back end)
```
vu.set_sim_option("ghdl.flags", ["-Wl,-lgcov"])
vu.set_sim_option("ghdl.flags", ["-Wl,--coverage"]) 
```
but the following simulation option (runtime option) does not:
`vu.set_sim_option("ghdl.flags", ["--assert-level=failure"])`
resulting in the message
"/usr/local/bin/ghdl: unknown option '--assert-level=failure' for command '--elab-run'

Command line usage of ghdl reveals:
```
ghdl --elab-run --std=08 --assert-level=none test
ghdl: unknown option '--assert-level=none' for command '--elab-run'

while
`ghdl --elab-run --std=08 test --assert-level=none`
works as expected.
```
I can work around this issue by changing the problematic asserts from severity Error (which abort the simulation) to severity Warning (which doesn't).

kraigher, 2016-07-22:
I guess we could add a `ghdl.elab_flags` sim_option as well

kraigher, 2016-07-22:
Also you should not that VUnit will always set --assert-level itself

Brian-Drummond, 2016-07-22:
ah, I wondered about the latter point : ghdl's default is documented to be Failure and my testbench was aborting on Error. 

How does VUnit set assert-level and is that controllable by the user? I didn't see any reference to that when I looked at the documentation. I presume it has internal reasons for doing so.

However it's slightly orthogonal to the issue fo whether options are passed in the right place on the command line, which is maybe worth solving anyway.

kraigher, 2016-07-22:
When we started VUnit we decided to set the default break assert level to error since that is what is the default of an assert statement. Thus an `assert false` would brake the simulation. There is no internal reason for having error as the default other than it has come to be expected by the users. It is possible to also break on warnings by putting the `-- vunit_pragma fail_on_warning` in the test bench. That seems to be missing in the documentation. There is not yet a public method to change the break level to fatal.

kraigher, 2016-07-22:
I split the sim_option ghdl.flags into ghdl.sim_flags and ghdl.elab_flags in the branch https://github.com/VUnit/vunit/tree/issue_185.

Brian-Drummond, 2016-07-22:
Thanks! Browsing the branch, it looks good...
Sorry to be such a pain, but having installed Vunit by "sudo pip install vunit", how does a Python newby go about updating my installation to this branch? 
```
git clone (think I can figure this bit out)
git checkout issue_185
some installation step?
```

kraigher, 2016-07-22:
`python setup.py install`. I typically do not install VUnit rather setting the PYTHONPATH environments variable to the repo root. It makes it easier to do development without overwriting system folders with installed python libraries.

Brian-Drummond, 2016-07-22:
Thanks : with "sudo", setup.py works. I'll bear in mind the alternative PYTHONPATH approach if I start working on vunit myself.

kraigher, 2016-08-07:
Merged the split flags now

closed 2016-08-07 by kraigher