VUnit/vunit #189: Improve language suport

leoheck, 2016-07-26:
Why not support other languages like:
- Verilog (probably icarus verilog is a good start)
- SystemC (gcc + systemc library)

cmarqu, 2016-07-26:
(System)Verilog(-AMS) at least parses fine, so a multi-language design with those languages is not a problem. SystemVerilog support for the testbench is there, just a bit more rudimentary than in VHDL.

Support for SystemC and Specman/e would be desirable indeed.

kraigher, 2016-10-06:
There are no plans by me to add support for SystemC or Specman/e and it is not likely to happen. Specman/e is not very interesting since it is a proprietary language for one specific simulator that I do not even have access to. SystemC could be more likely but I have so far not seen much interest. Personally I do not think SystemC is a good technical solution so I am not very interested in it. I could imagine adding some compile / link support for SystemC if it is not very hard and some user is willing to invest the time. One could probably work with SystemC dependencies today by pre-compiling them and adding custom simulator flags.

closed 2016-10-06 by kraigher