VUnit/vunit #194: Modify instance/{lib,architecture} bindings in Python Interface

RafaelCatrou, 2016-08-17:
VHDL configuration are often used to modify the bindings between instances and library.entity(architecture) ; mostly for switching between functional architectures and stubs.

In the VUnit python interface, the `add_config() `method of `class vunit.ui.TestBench` allows to easily play with generics.

Question: Is there a way to modify architecture+lib bindings too?
If not, it would be a nice to have!

joshrsmith, 2016-08-18:
This is similar to #179 I believe

RafaelCatrou, 2016-08-19:
Thanks Josh, I agree it revolves around a similar issue: configuration.
The point I'd like to raise is not necessarily about the complete support of VHDL configuration files itself (for example: `.add_source(my_configuration.vhd)` ), which might sometimes be complex.
It's more about the ability to select architecture of a submodule using a python method (or else), which seems to be more basic to implement.
But if the #179 is implemented, I would be ok too (who can do more can do less, after all ).

joshrsmith, 2016-08-20:
How would you normally do this with your simulator, without using a configuration?

RafaelCatrou, 2016-08-20:
I meant to rely on configuration generated by VUnit, configured using Python interface (like for  generics).
But it raises a question: when, in run.py, the following kind of code is used to play with generics : 
`for my_generic_boolean in [False, True]:`
 &nbsp; &nbsp; &nbsp; `tb.add_config(..., generics=dict(my_generic_bool=my_generic_bool), ...)`
At first, I thinked that it causes VUnit to generate automatically 2 temporary VHDL configurations files (one with True, one with False), then run each of them one after the other (GHDL supporting VHDL configuration file).
But now rethinking to the replies in the other issue I'm starting to doubt. Can you shed some light on this please? Where can I find a presentation of VUnit internal mecanisms (to help me before start reading the code) ?

joshrsmith, 2016-08-21:
Currently VUnit does not use configurations as you know them in the VHDL language.

A "configuration" in VUnit is a way that a testbench can be configured to run, and typically all test cases are run for each configuration specified. A VUnit configuration does not generate VHDL, it only indicates things like -- how test files should be prepared (`pre_config`), any generics that should be set (`generics`), and anything that should be done after the testcase runs (`post_check`).

So in your example, by doing `tb.add_config`, you are adding a configuration for VUnit to test (which replaces the default). Internally VUnit will set the top level of simulation as the testbench entity (e.g. `my_module_tb`) and set its corresponding generic(s) that you define in the dict. All test cases that are part of the testbench architecture of `my_module_tb` will be simulated using the generic that was specified. If more configurations are added through `add_config`, then all test cases will be simulated for those configurations as well.

RafaelCatrou, 2016-08-21:
This is crystal clear! Thank you Josh.
As I need to modify architecture bindings of DUT in my tests, I workaround this issue by just using direct instantiation of VHDL configurations in my testbench. Whereas this is static hierarchical stuff, this does the job for my needs (as generics like `my_generic_boolean` coming from the Python script in the example above is passed to the DUT. I don't pass the generic "runner_cfg : string" to the DUT  as only the testbench makes use of it).
`run.py -> my_module_tb -> my_module_config -> my_module`
It works as expected.

We can close this issue

closed 2016-08-22 by RafaelCatrou