VUnit/vunit #195: Question on class vunit.ui.TestBench.add_config() -> pre_config

RafaelCatrou, 2016-08-19:
The [manual][1] says for the parameter `pre_config` of the  `class vunit.ui.TestBench.add_config()` method that 

>  class vunit.ui.TestBench
>
>    User interface of a test bench. A test bench consists of one or more >Test cases. Setting options for a test bench will apply that option all test >cases belonging to that test bench.
>
>    add_config(name='', generics=None, parameters=None, pre_config=None, post_check=None)
>
>Add a configuration of this test bench. Multiple configuration may be added one after another. If no configurations are added the default configuration is used.
>Parameters:	
>
>     name – The name of the configuration. Will be added as a suffix on the test name
>     generics – A dict containing the generics to be set
>     parameters – A dict containing the parameters to be set
>     pre_config – A function to be called before test execution. The function may accept a string which is the filesystem path to the directory where test outputs are stored. The function must return True or the test will fail
>     post_check – A function to be called after test execution. The function must accept a string which is the filesystem path to the directory where test outputs are stored. The function must return True or the test will fail

Difference between `pre_config` and `post_check` lies in the **may/must** *accept a string which is the filesystem path to the directory where test outputs are stored*.

Whereas `post_check` is called with the path as expected, I can't figure how to get `pre_config` being called with the path too. Is it working?

The function assigned to `pre_config` and `post_check` are similar to the model:

    my_method(self, path_of_run="" ):

But in the case of `pre_config`, the variable path_of_run remains empty . Why?

Thanks for your help

  [1]: https://vunit.github.io/python_interface.html

joshrsmith, 2016-08-20:
I think I updated the documentation for `pre_config` awhile ago, and it looks like I was mistaken.

That said, I think it _should_ receive the path where the test outputs are stored. That way you can generate test inputs (e.g. a stimulus file to be read by the testbench) in a location that is specific to the test.

@kraigher Any thoughts on this?

joshrsmith, 2016-08-20:
Actually, it looks like it does accept the path.
See: https://github.com/VUnit/vunit/blob/7097fc87efac668b0262173d944d8e2347dfc8a5/vunit/test_suites.py#L223

I think your issue may be your argument is not named `output_path`
This is an issue because of the way the code inspects the function signature to check if it is expecting to receive this argument. I am guessing if you change your pre_config to accept `output_path` it will work correctly.

kraigher, 2016-08-20:
Yes it must have a single argument named output_path

RafaelCatrou, 2016-08-20:
I changed to `my_method(self, output_path="" ):` and it works like a charm :+1: 
Thanks much!

closed 2016-08-20 by RafaelCatrou