VUnit/vunit #198: Incisive simulations never pass, even though they should

adrspa, 2016-09-29:
Simulations don't exit gracefully but always fail (even if they should in fact pass). This is what I get at the end of a simulation:

REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Simulation completed successfully!
ASSERT/FAILURE (time 4593901 PS) from procedure @vunit_lib.vunit_stop_pkg:vunit_stop
Assertion violation.
Assertion at 4593901 PS + 4
/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit/vunit/vhdl/core/src/stop_body_93.vhd:10 assert false severity failure;
ncsim> catch {if {#vunit_pkg::runner.exit_without_errors == 1} {exit 0} else {exit 42}}
1
ncsim> catch {if {#run_base_pkg.runner.exit_without_errors == "TRUE"} {exit 0} else {exit 42}}
1
ncsim> exit
fail (P=0 S=0 F=1 T=1) lib.tb_zcom_responder.tc_axi_rw_check (3.7 seconds)
Summary
fail lib.tb_zcom_responder.tc_axi_rw_check (3.7 seconds)
pass 0 of 1
fail 1 of 1
Total time was 3.7 seconds
Elapsed time was 3.7 seconds
Some failed!

The "Simulation completed successfully!" is reported by my TB and after that the test_runner_cleanup(runner) is executed. See code below:

```
REPORT "Simulation completed successfully!" SEVERITY note;
test_runner_cleanup(runner);
WAIT;
```

Could this be related to this warning that I get:
ncelab: *W,EVGNTP: generic not of an appropriate type (tb_zcom_responder.runner_cfg).

LarsAsplund, 2016-09-30:
I think we've already solved the warning issue on gitter but I'll post it here for the record. Incisive cannot handle the `runner_cfg_t` type for the `runner_cfg` generic. Use `string` instead (`runner_cfg_t` is a subtype of `string`).

After your report statement I would like you to do `report to_string(get_checker_stat));` Does that show zero failing checks?

cmarqu, 2016-10-05:
VUnit deprecated usage of ``runner_cfg_t`` (because of Incisive) some months back (in https://github.com/VUnit/vunit/commit/6916ee5ebee4b212564fbd75d52d14c7c948d31b), so @adrspa , do you maybe have some old code lying around?

adrspa, 2016-10-06:
@LarsAsplund No, it does not. This is what I get from that report statement:
`REPORT/NOTE (time 329589 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Checks: 5
Passed: 0
Failed: 5`

So it would seem that something is wrong here. What else can I check?

@cmarqu I switched types on the runner_cfg generic to a string after @LarsAsplund commented on that on gitter. That has not helped with this issue however.

LarsAsplund, 2016-10-06:
@adrspa Do you agree that there are 5 checks in your testbench? Do you get any error messages from those if you run your script with the `-v` option? It might be something with the _stop level_. The default behavior is to stop a simulation at a failing check but it is possible to proceed such that you reach the `test_runner_cleanup` call (and your "Simulation completed successfully" report statement). VUnit will still keep track on all errors and fail if there are any . Is it possible to see your testbench? At least everything related to VUnit?  You can also add another report statement at the end:

``` vhdl
    report "Stop level is : " & log_level_t'image(default_checker.logger.log_stop_level);
```

Are you doing any calls to `checker_init`? Also add this report statement before the call to `test_runner_setup` 

``` vhdl
    report "Has active Python runner: " & boolean'image(active_python_runner(runner_cfg));
```

adrspa, 2016-11-10:
@LarsAsplund Sorry for the horrendously late reply but I've been on vacation for a month. Anyway, I blew the dust off of my TB and added the two report statements you suggested above. After deleting the vunit_out folder and kicking off a TC in batch mode this is what I got after all the compilation messages:

irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncelab: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
	Top level design units:
		:tb_zcom_responder:
ncvhdl_cg: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncsim: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncsim> run
REPORT/NOTE (time 0 FS) from process :main (architecture lib.tb_zcom_responder:beh)
Has active Python runner: true
Corrupt frozen dictionary item "output pa" in "active python runner : true,enabled_test_cases : tc_axi_rw_check,output pa"d½-
                                                                                                                              Æ-
                                                                                                                                 ôÞ/
                                                                                                                                    ¨½-
                                                                                                                                       º-
  ÔÞ/
     º-
        àÞ/
           <½-
              èÞ/
                 d½-
                    $Ú-
                       ñ rN	ÐÊ0
                                   º-
                                      ¼Þ/

                                         ß/
                                           º".
Corrupt frozen dictionary item "output pa" in "active python runner : true,enabled_test_cases : tc_axi_rw_check,output pa"d½-
                                                                                                                              Æ-
                                                                                                                                 ôÞ/
                                                                                                                                    ¨½-
                                                                                                                                       º-
  ÔÞ/
     º-
        àÞ/
           <½-
              èÞ/
                 d½-
                    $Ú-
                       ñ rN	ÐÊ0
                                   º-
                                      ¼Þ/

                                         ß/
                                           º".
[... cut ...]
I've seen the error about the "Multiple logical library mappings" before and haven't really bothered with them. No idea if they are caused by VUnit, or if it is our setup of the Cadence tools that's doing it. 

Has active Python runner: true - is that what you wanted to see?

The "frozen dictionary" errors are a bit concerning. Ideas?

5 total checks where all fail. What checks does VUnit refer to here? 

Stop level is : error

But if I run the same simulation again (without deleting the vunit_out folder first) I instead get a different result. See next post.

adrspa, 2016-11-10:
Starting lib.tb_zcom_responder.tc_axi_rw_check
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncelab: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
	Top level design units:
		:tb_zcom_responder:
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
irun: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncsim: *W,DLNCML: Multiple logical library mappings have been detected in the cds.lib. Directory used multiple times '/afs/sunrise.ericsson.se/se/app/vbuild/RHEL6-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS'.
ncsim> run
REPORT/NOTE (time 0 FS) from process :main (architecture lib.tb_zcom_responder:beh)
Has active Python runner: false
REPORT/NOTE (time 0 FS) from process :main (architecture lib.tb_zcom_responder:beh)
This is TC AXI RW check!
     ---  AXI_TX_STIMULI: # tc_axi_rw_check
     ---  AXI_TX_STIMULI: # Set tb print mode
     ---  AXI_TX_STIMULI: TB print mode set to: VERBOSE
     ---  AXI_TX_STIMULI: # Initial wait
     ---  AXI_TX_STIMULI: WAIT: 0x00000200
     ---  AXI_TX_STIMULI: # 
     ---  AXI_TX_STIMULI: # End simulation
     ---  AXI_TX_STIMULI: FORCE SIGNAL
     ---       ---  ^.s_axi_stim_complete  true
ASSERT/NOTE (time 4439279 PS) from process :tb_tc_decoder_i0:stimuli_proc (architecture lib.tb_tc_decoder:bhv)
AXI stimuli file ended...
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Simulation completed successfully!
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Checks: 0
Passed: 0
Failed: 0
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Stop level is : failure
ASSERT/FAILURE (time 4593901 PS) from procedure @vunit_lib.vunit_stop_pkg:vunit_stop
Assertion violation.
Assertion at 4593901 PS + 4
/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit/vunit/vhdl/core/src/stop_body_93.vhd:10     assert false severity failure;
ncsim> catch {if {#vunit_pkg::__runner__.exit_without_errors == 1} {exit 0} else {exit 42}}
1
ncsim> catch {if {#run_base_pkg.runner.exit_without_errors == "TRUE"} {exit 0} else {exit 42}}
1
ncsim> exit
fail (P=0 S=0 F=1 T=1) lib.tb_zcom_responder.tc_axi_rw_check (3.2 seconds)

==== Summary =================================================
fail lib.tb_zcom_responder.tc_axi_rw_check (3.2 seconds)
==============================================================
pass 0 of 1
fail 1 of 1
==============================================================
Total time was 3.2 seconds
Elapsed time was 3.2 seconds
==============================================================
Some failed!

Still the same "Multiple logical library mappings" error.

The corrupt dictionary errors are mysteriously gone now.

Has active Python runner: false - this has changed!

Checks 0 - this has also changed!

Stop level is : failure - and so has this.

adrspa, 2016-11-10:
I noticed that if I keep starting the same simulation over and over it keeps ping-ponging between these two outputs. 

Is there anything else I can check?

adrspa, 2016-11-10:
Actually, the results don't exactly ping-pong. I ran a few one after the other (always starting the second TC which is the tc_axi_rw_check) and the results vary. I've attached a textfile containing the results. 

Note that sometimes the first TC (tc_hello_world) is executed instead, but a separate generic that points to a AXI command file always points to the correct file. Strange. 
[tb_zcom_responder_runs.txt](https://github.com/VUnit/vunit/files/583449/tb_zcom_responder_runs.txt)

kraigher, 2016-11-10:
The corrupt frozen dictionary items indicate that the `runner_cfg` generic is truncated or corrupt in some other way. The source of randomness that you experience could be that the `runner_cfg` is not created deterministically due to Python dictionary implementation where iterating over the keys of the dictionary has no defined order and can be random. This randomness can cause different parts of the runner_cfg to be affected by the truncation causing different test bench behavior. I will push a branch where I remove this randomness and also adds a report directly in test_runner_setup which prints the runner_cfg and its length.

kraigher, 2016-11-10:
@adrspa I pushed the deterministic `runner_cfg` creation and printout of `runner_cfg` in `test_runner_setup` to the `incisive_investigation` branch. You can try it out and see if it provides us with any valuable information.

adrspa, 2016-11-14:
@kraigher I removed vunit_out and then executed tc_axi_rw_check. This is what I got after the compilation was completed:

run
REPORT/NOTE (time 0 FS) from process :main (architecture lib.tb_zcom_responder:beh)
Has active Python runner: true
REPORT/NOTE (time 0 FS) from procedure @vunit_lib.run_pkg:test_runner_setup
runner_cfg'length = 191
REPORT/NOTE (time 0 FS) from procedure @vunit_lib.run_pkg:test_runner_setup
active python runner : true,enabled_test_cases : tc_axi_rw_check,output pa
Corrupt frozen dictionary item "output pa" in "active python runner : true,enabled_test_cases : tc_axi_rw_check,output pa"d¬R
µR
tÚìU	S
©R
TÚìU©R
`ÚìU<¬R
hÚìUd¬R
(èR
ñ rN	õîU©R
<ÚìUÚìU©R
ÚìU<¬R".

-- lots of these "Corrupt frozen dictionary item" errors

REPORT/NOTE (time 0 FS) from process :main (architecture lib.tb_zcom_responder:beh)
This is TC AXI RW check!
     ---  AXI_TX_STIMULI: # tc_axi_rw_check
     ---  AXI_TX_STIMULI: # Set tb print mode
     ---  AXI_TX_STIMULI: TB print mode set to: VERBOSE
     ---  AXI_TX_STIMULI: # Initial wait
     ---  AXI_TX_STIMULI: WAIT: 0x00000200
     ---  AXI_TX_STIMULI: # 
     ---  AXI_TX_STIMULI: # End simulation
     ---  AXI_TX_STIMULI: FORCE SIGNAL
     ---       ---  ^.s_axi_stim_complete  true
ASSERT/NOTE (time 4439279 PS) from process :tb_tc_decoder_i0:stimuli_proc (architecture lib.tb_tc_decoder:bhv)
AXI stimuli file ended...
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Simulation completed successfully!
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Checks: 5
Passed: 0
Failed: 5
REPORT/NOTE (time 4593901 PS) from process :main (architecture lib.tb_zcom_responder:beh)
Stop level is : error
ASSERT/FAILURE (time 4593901 PS) from procedure @vunit_lib.vunit_stop_pkg:vunit_stop
Assertion violation.
Assertion at 4593901 PS + 4
/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit/vunit/vhdl/core/src/stop_body_93.vhd:10     assert false severity failure;
ncsim> catch {if {#vunit_pkg::__runner__.exit_without_errors == 1} {exit 0} else {exit 42}}
1
ncsim> catch {if {#run_base_pkg.runner.exit_without_errors == "TRUE"} {exit 0} else {exit 42}}
1
ncsim> exit
fail (P=0 S=0 F=1 T=1) lib.tb_zcom_responder.tc_axi_rw_check (3.3 seconds)

==== Summary =================================================
fail lib.tb_zcom_responder.tc_axi_rw_check (3.3 seconds)
==============================================================
pass 0 of 1
fail 1 of 1
==============================================================
Total time was 3.3 seconds
Elapsed time was 3.3 seconds
==============================================================
Some failed!

kraigher, 2016-11-14:
@adrspa It obviously looks like the `runner_cfg` generic was truncated.

kraigher, 2016-11-14:
Could you paste the contents of the file named `irun_simulate.args` that is written to the test case output folder by `incisive_interface.py`, that way I can see what the `output_path` is supposed to be. We need to understand why incisive truncates the string.

adrspa, 2016-11-15:
@kraigher Sure thing! I found the file here:
./vunit_out/test_output/d93401681993690c9a0eff5fe0efe325a673f416/irun_simulate.args

And this is what's in it:

-nocopyright
-licqueue
-errormax 10
-nowarn WRMNZD
-nowarn DLCPTH
-nowarn DLCVAR
-ncerror EVBBOL
-ncerror EVBSTR
-ncerror EVBNAT
-work work
-nclibdirname "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/libraries"
-access +rw
-cdslib "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/cds.lib"
-log "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/test_output/d93401681993690c9a0eff5fe0efe325a673f416/irun_simulate.log"
-quiet
-gpg "tb_zcom_responder.runner_cfg => \"active python runner : true,enabled_test_cases : tc_axi_rw_check,output path : /vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/test_output/d93401681993690c9a0eff5fe0efe325a673f416/\""
-gpg "tb_zcom_responder.g_axi_stim_file => \"/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/axi_stimuli/tc_axi_rw_check.txt\""
-reflib "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/libraries/vunit_lib"
-reflib "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/libraries/lib"
-reflib "/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/libraries/fifo_generator_v13_1_1"
-reflib "/app/vbuild/SLED11-x86_64/cadence_incisive/15.10.011/tools.lnx86/inca/files/NCUTILS"
-access +r
-input "@run"
-input "@catch {if {#vunit_pkg::__runner__.exit_without_errors == 1} {exit 0} else {exit 42}}"
-input "@catch {if {#run_base_pkg.runner.exit_without_errors == \"TRUE\"} {exit 0} else {exit 42}}"
-top lib.tb_zcom_responder:beh

And thanks for all the help!

cmarqu, 2016-11-15:
Strange. That args file looks fine to me.
You can use the option ``--log-level debug`` to ``run.py`` to get some more output, but that will just show you that it invokes ``irun -f irun_simulate.args`` (for each of the directories under ``vunit_out/test_output/``).
Maybe try doing a ``cd`` to such a directory, then run ``irun -f irun_elaborate.args`` and ``irun -f irun_simulate.args`` by hand. This alone will not change much, but you can then add some more options to that command line (see ``irun -helpall``). For instance, you could try adding ``-64bit``, or ``-gverbose``.  
Also, what does ``echo $CDS_AUTO_64BIT`` give you?

cmarqu, 2016-11-15:
Actually, yes, try ``irun -gverbose -f irun_elaborate.args`` please. This should print something with "``assigned a value``".

kraigher, 2016-11-15:
@adrspa The runner_cfg string is truncated at the exact same length as the `runner_cfg_default` constant.

```
# runner_cfg_default:
enabled_test_cases : __all__, output path : , active python runner : false

# your truncated runner_cfg:
active python runner : true,enabled_test_cases : tc_axi_rw_check,output pa
```

I would suggest you remove the default value of the `runner_cfg` generic to try to avoid this problem. I think it was @cmarqu who discovered that Incisive will you allow you to provide a longer string generic than the one provided by the default argument. It is crappy behavior from Incisive to be sure, it seems to set the string length to the right value tough since the printout gave 191.

@LarsAsplund can we document this limitation in a better way?

cmarqu, 2016-11-15:
Oh. I always was under the impression that this change was in since a long time...
VUnit already has https://github.com/VUnit/vunit/blob/master/incisive_vhdl_fixup.py, maybe it could be added there if removing the default value has other side effects.

kraigher, 2016-11-15:
@cmarqu We already changed our example test benches. The default value is within his test bench. That is not something we can change for him.

cmarqu, 2016-11-15:
@kraigher I see, so @adrspa probably used a testbench from before that change as a reference.
@adrspa It may be worth filing a bug with Cadence. (I might have done this already, but cannot check because I changed companies since. Anyway, the more tickets the merrier :)

LarsAsplund, 2016-11-15:
@kraigher What we're are lacking is a complete documentation of the `run_pkg` package. All we have is the user guide covering part of its functionality. People are asking question that can be solved by its hidden features so it has been on my mental todo list for a while.

Meanwhile, can't we add a warning if the test scanner finds an init value on the `runner_cfg` generic?

joshrsmith, 2016-11-16:
@LarsAsplund What is wrong with an init/default value on `runner_cfg` generic? Isn't this necessary (if you don't want to specify the long string on the simulation invocation on the command line) in order to run the simulation directly via your simulator (instead of through VUnit/python interface)?

adrspa, 2016-11-16:
@kraigher @cmarqu I removed the default on the runner_cfg in my TB and that solved the problem for me. Now my two TC's run and then exit gracefully. Thanks!

closed 2016-11-16 by adrspa