VUnit/vunit #199: Incisive - unable to re-compile with VUnit with a simulation open in GUI mode

adrspa, 2016-09-29:
When trying to re-compile with VUnit while a simulation is open in GUI mode I get this error:

Compiling /vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/rtl/zcom_responder_decoder.vhd into lib ...
irun: *W,WKWTLK: Waiting for a Exclusive lock on file '/vobs/rbs/hw/ru_fpga/test_boards/ru_lib/zcom/sim/vunit_out/incisive/libraries/irun.lnx86.15.10.nc/.nclib.lock'. pid:4209.

The only way around it seems to be to close the simulator and then restart it after compilation. This is of course very tedious when only doing small changes during debugging.

joshrsmith, 2016-09-30:
Work-around could be possible if #181 is solved in a simulator independent manner

adrspa, 2016-10-06:
@joshrsmith Yes, I suppose that would suffice. Unfortunately I'm a VHDL hacker and not well versed enough in python to fix that myself...

kraigher, 2016-10-06:
@adrspa Hard for me to help you here since I do not have access to Incisive. It seems you need to figure out what makes incisive place that lock when in GUI mode and see if you can release it somehow. 

@joshrsmith Even if #181 has a simulator independent solution Incisive will maybe still make lock files that inhibit re-compilation when inside the GUI.

adrspa, 2016-11-15:
@kraigher All right, I'll see if I can figure that out and then I'll get back here.

kraigher, 2017-07-19:
@adrspa Any news on this front?

adrspa, 2017-08-16:
Hi,
No, not really. The problem remains but I haven't had a chance to devote more time to it, especially since we switched back to Questa. 

Thanks for checking in on this. I'll post here if we solve this problem in the future.

adrspa, 2017-08-16:
I'm closing this issue as it is probably related to Incisive rather than VUnit itself.

closed 2017-08-16 by adrspa