VUnit/vunit #203: log_types.vhd - possible bug

pbrown76, 2016-10-28:
When compiling vunit/vhdl/logging/src/log_types.vhd, Modelsim gives a warning on line 25 "Range 1 to 0 is null". Looking at the source, line 25 declares a constant of type log_level_vector_t. The log_level_vector_t is a natural array. Natural is 0 to integer'HIGH, so line 25 should be log_level_vector_t(0 to 1), instead of "1 to 0". When I make the change, the compiler warning goes away.

LarsAsplund, 2016-10-28:
@pbrown76 Sometimes it is convenient to specify empty arrays of length 0 and that's what you get with a 1 to 0 range. Rather than defining these empty arrays in several places and get several compiler warnings I have a single constant which is used in a number of placed internally in VUnit.

closed 2016-10-28 by LarsAsplund

mitchsm, 2016-10-28:
FWIW, you can avoid the compiler warning by using the range of a null-length string:

```VHDL
constant c_NULL : string := "";
constant example_empty_array : integer_vector(c_NULL'range) := (others => 0);
```

LarsAsplund, 2016-10-28:
@mitchsm Thanks for the tip, I'll try that and see if it "fools" all of the supported simulators.

LarsAsplund, 2016-10-28:
@mitchsm GHDL was ok with the previous solution but none of Active-HDL, Riviera-PRO, and ModelSim. With the proposed change ModelSim passes without warnings but Active-HDL and Riviera-PRO are still suspicious.

joshrsmith, 2016-10-28:
Aldec's products seem picky about everything!