VUnit/vunit #204: IOError due to long path to .vunit_hash file

bleviet, 2016-11-03:
Hi,
while executing VUnit on Windows and Python 2.7.12, the following error occurs with very long path names (for example in my case 296 characters)

> File "C:\Python27\lib\site-packages\vunit_hdl-0.71.0-py2.7.egg\vunit\project.p
y", line 399, in update
    ostools.write_file(self._hash_file_name_of(source_file), new_content_hash)
  File "C:\Python27\lib\site-packages\vunit_hdl-0.71.0-py2.7.egg\vunit\ostools.p
y", line 269, in write_file
    with open(file_name, "w") as file_to_write:
IOError: [Errno 2] No such file or directory: '<long_path_with_296_characters>.vunit_hash'

I've found a solution for that by changing the line 399 in project.py:

> ostools.write_file(self._hash_file_name_of(source_file), new_content_hash)

to

> ostools.write_file("\\\\\\\\?\\\\" + self._hash_file_name_of(source_file), new_content_hash)

(Link to this solution: http://stackoverflow.com/questions/14075465/copy-a-file-with-a-too-long-path-to-another-directory-in-python)

Please verify and provide an appropriate solution for that. Thank you!

Best regards,
Bach

kraigher, 2016-11-03:
The root cause of this is that the output path or external library path has been set to a path that is to long given the windows path length requirement. Are you sure that this is the only problem that arises when having too long path names? Will the simulators not also fail to handle this?

bleviet, 2016-11-04:
The path to the file to be compiled has less than 256 characters, so the simulator is fine with it. Here VUnit is trying to write a .vunit_hash file. I don't know which purpose this hash file has (to track changes for recompilation?), would you please explain?

kraigher, 2016-11-05:
The purpose of the hash file is to know when to recompile by tracking the contents of the file the last time it was compiled.

I still believe you will run into problems with long paths even with adding the path prefix above. The error implies that the `--output-path` is some path with a length close to 256. All simulator logs and other output artifacts are stored in the output path. Have you tried simulating anything with this long path?

bleviet, 2016-11-07:
In my situation after adding the path prefix, the compilation process gone through and Modelsim started. However some errors occur while Modelsim is trying to loading the design. I need to make more investigation in this and will report.

bleviet, 2016-11-08:
After fixing the errors I mentioned, which have no relationship with IOError, the simulation runs and provides expected results. So I can confirm in this case, the IOError is solved by adding the prefix "\\\\\\\\?\\\\".

kraigher, 2016-11-08:
I am not convinced this is something that should be "fixed" in VUnit. The line you list is just one example where we read or write files. I am not keen on just fixing this in one place without going all the way. There will be many places where this needs to be fixed. To handle long paths on Windows all places where we access files would have to be an absolute path with //?/ prefix. Since we also support Linux we would have to wrap all file_name operations into some function that either adds //?/ or not depends on operating system. Most Python libraries and even the Python standard libraries (https://bugs.python.org/issue18199) do not perform such work arounds on Windows. Any Windows user must live with the path limitation, it is a flaw in the operating system design. As Windows user you must thus avoid creating long paths. This is true when using any Windows software and also affects VUnit.

closed 2016-11-08 by kraigher