VUnit/vunit #209: Add support for Vivado XSIM

LarsAsplund, 2016-11-17:
We have started to do some work but we didn't have an issue for it so I'm creating one.

flamebut57, 2017-04-07:
Any update on this? Any help needed?

LarsAsplund, 2017-04-13:
@flamebut57 This issue has been dormant awaiting more interest. Maybe we're seeing that now? There were also some tool issues the last time we made some prototyping:
* XSIM didn't support custom resolved types which VUnit uses. This can be worked around but it would be easier if we didn't have to
* XSIM didn't support setting the severity level at which the simulation stops

We can surely use some help. Interested? Finding out if these two issues remains would be a starting point.

drizzd, 2017-08-25:
I just tested both issues with Vivado 2013.1, 2016.2 and 2017.2.

- I didn't manage to compile run_types.vhd directly due to the large number of dependencies, but I could compile my own user defined resolved type just fine.
- I could find no option to set the severity level at which the simulation stops. Xsim will continue on severity warning or severity error. It will stop on severity failure. At that point the user can continue the simulation by writing run -all to standard input. Do we need to stop on warning or error?

LarsAsplund, 2017-08-26:
@flamebut57 Is this true with all the tested versions? I can't remember what version I used to test user defined resolved types but 2013.1 should have been available at the time. What did you type look like? Was it based on `std_logic`?

aodling, 2017-10-06:
Hi!

Any updates to this issue?

kraigher, 2017-10-10:
There is no active work to add support for xsim. There have been some preliminary tests which shows it lacks enough VHDL support of even 93 standard to accept the VUnit code. Also it has a very sluggish startup time which would hinder fast feedback when running multiple small tests. The first issue can be worked arround in VUnit at least but takes effort which I am currently not willing to spend. If anyone in the community wants to give it a go fine by me.

pedronevestopic, 2017-11-15:
I am not very experienced in Python, but I am in VHDL. I have some time and I would like to look into this to see if I can make it work with XSIM. Where should I start from? Is there a branch of previous attempts?

kraigher, 2017-11-15:
@pedronevestopic I pushed a branch with a work in progress implementation of Python-level support. I do not know how well it works any more. The major blocker is the lack of VHDL support in particular user defined resolved types. Those limitations could be worked around though.

jklapel, 2018-02-13:
@pedronevestopic Were you able to make any progress on making the VHDL work in XSIM?

schluta, 2018-02-13:
I apologize as I have been very busy in the last year. I did throw together
a template to use Vivado Simulator for both Verilog and VHDL. I was
attempting to run the instance of Vivado in a VM and I am unsure if I was
running into memory issues but if I remember correctly it just hung
forever.

To clarify I believe the VHDL test stated that there was an unsupported
feature ( it was a compile order issue where it stated one file changed
after another compiled this may have changed in newer versions I tested on
2015.4 and 2016.4), while the Verilog test just hung forever.

I would be happy to share these templates and test cases that I added if
they are of interest or use to the community.

Let me know if there is any interest.

Thanks,

PetterssonMagnus, 2018-02-25:
I am interested in xsim support, although from a purely (system)verilog perspective. Would https://github.com/VUnit/vunit/commit/1f91ea08f7f8e56b334116374d6676544b737858 be a good starting point should I have some spare time over?

pedronevestopic, 2018-02-25:
I’m sorry. I couldn’t do anything . I got a problem on my back and going to need a surgery. I haven’t been able to sit on a computer. If no one has spend any time when I’m recovered in some months , I’ll pick it up.

kraigher, 2018-02-26:
@PetterssonMagnus The xsim branch contains a work in progress that you can experiment with

jklapel, 2018-02-26:
@PetterssonMagnus I was able to take the xsim branch and get it working for VHDL testbenches at a very basic level with only a day or so of tinkering with it.  Getting Verilog to work with xsim shouldn't be too much different I don't believe, though I haven't tried it.  I ran into bigger issues with the custom resolved types in VHDL.  (Take a look at the conversation I had with kraigher on February 13th on the Gitter chat.)  I don't know how the Verilog side of VUnit works, but my understanding is that Xilinx doesn't have a lot of support SystemVerilog.

PetterssonMagnus, 2018-02-26:
Ok, I have been able to successfully get the example user_guide through xvlog thus creating libraries vunit_lib and lib. Subsequently xelab seems to lock up, neither finish executing nor providing me with any error messages.

`xelab -L vunit_lib -L lib --runall lib.tb_example --generic_top "runner_cfg=active python runner : true,enabled_test_cases : Test that a test case that takes too long time fails with a timeout,output path : C::/repo/vunit/examples/verilog/user_guide/vunit_out/test_output/lib.tb_example.f763c9dbbd826025101721b40758dfe8d596eff6/" -v 1 -O0 -mt off
`
I get the following output:
....
Starting static elaboration
INFO: [VRFC 10-520] compiling module tb_example [C:/repo/vunit/examples/verilog/user_guide/tb_example.sv:12]
Completed static elaboration
Starting simulation data flow analysis
Completed simulation data flow analysis
SDG Object Count: 375, SDG Object Memory Usage: 46 KB.
Time Resolution for simulation is 1ps
Compiling module lib.tb_example

... and then the fun ends ... id est nothing more happens.

LarsAsplund, 2018-02-26:
@PetterssonMagnus What happens if you setup the project from within XSIM? Does it fail? Does it give you  more info? If not, I would start doing some divide an conquer

PetterssonMagnus, 2018-02-28:
Eventually I got verilog examples "uart" and "user_guide" to work in non-gui mode. Have not verified, but I doubt xsim supports verilog-ams. Created a fork: https://github.com/PetterssonMagnus/vunit/tree/xsim .

LarsAsplund, 2018-03-11:
@jklapel @kraigher @PetterssonMagnus I have pushed a VUnit version without custom types to a new [branch](https://github.com/VUnit/vunit/tree/remove-custom-resolved-types). I'm keeping it on a branch because I need someone to test if Incisive still works. The `runner` signal which used to be of a custom resolved type is currently a two bit `std_logic_vector` but I may take that another step and make it a single `std_logic`. That work wouldn't affect the porting to XSIM much so you can go ahead with what I have now.

LarsAsplund, 2018-07-29:
@jklapel @kraigher @PetterssonMagnus I've merged the aforementioned branch to master now such that it doesn't use any custom resolved types.

haggaie, 2019-05-31:
Hi, I rebased the xsim support on master and tested the SystemVerilog examples. I had to modify the code a bit to make it work on Linux.

LarsAsplund, 2019-05-31:
I'll test this. I think it's about time we get this going so any help is appreciated. @PetterssonMagnus Do you have any comments on the status of your work?

PetterssonMagnus, 2019-05-31:
Hm, this was a while ago.
Status, Been using vunit and xsim for sv testbenches. The only non pushed, functional, addition I have is a crude way of specifying Xsim initfile. Used to locate ips and other Xilinx libraries.
At some point I had the vhdl examples compiling but ran into null pointer exceptions immediately at startup(elab). Think it was due to the order of objects (runner perhaps) being created / referenced. The vunit internal vhdl code required a few hours of transformations to get to that.

ro-ca, 2019-07-10:
Hi! Is there any progress on this? What's the best way to test out what y'all have or help out? Still new to GitHub, but good at python / vhdl. Looks to my untrained eye that @haggaie and @PetterssonMagnus 's branches are identical, but both may have fallen somewhat behind the master. Is that accurate? Any thoughts on what needs to be done to get vhdl compiling?

1138-4EB, 2019-07-10:
@ro-ca, I rebased their branch on top of master, so you can try it: https://github.com/1138-4EB/vunit/tree/petterssonmagnus-xsim

imd1, 2019-10-24:
Hi 

I've downloaded  https://github.com/1138-4EB/vunit/tree/petterssonmagnus-xsim and tried to run one of my existing testsuites (OK under Questasim and GHDL) using Vivado 2018.3. and I get the following compile error for a VUnit library file:

ERROR: [VRFC 10-1261] prefix of LENGTH attribute should be an array or array (sub)type [/usr/lib/....../vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:93]

Is this expected, anticipated? Are there any workarounds?

Regards
Ian

1138-4EB, 2019-10-24:
It seems to be an unsupported feature in Vivado.

- The *prefix* is `ptrs(ref)`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd#L93).
- `ptr` is type `vava_t`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd#L58.
- `vava_t` is an alias of `string_access_vector_access_t`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/string_ptr_pkg.vhd#L27.
- `string_access_vector_access_t` is type `access string_access_vector_t`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/types.vhd#L12.
- `string_access_vector_t` is an array of `string_access_t`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/types.vhd#L11.
- `string_access_t` is type `access string`: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/types.vhd#L10.

Therefore, `ptrs(ref)` is/returns an `access` to a `string`. It might be possible to work around the issue by rewriting function `length`:

```vhdl
    impure function length (
      ref : natural
    ) return integer is
    variable s : string_access_t := ptrs(ref);
    begin
      return s'length;
    end;
```

Further notes:

- This will probably need to be *fixed* in other places too. For example: https://github.com/1138-4EB/vunit/blob/petterssonmagnus-xsim/vunit/vhdl/data_types/src/integer_vector_ptr_pkg-body-2002p.vhd#L85
- https://github.com/1138-4EB/vunit/tree/petterssonmagnus-xsim has not been rebased after merging #507 into master. These sources were modified in #507: https://github.com/VUnit/vunit/blob/master/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd#L191-L200
  - Will you keep a working copy of the branch in your fork?
  - Are you ok with fixing the current status of https://github.com/1138-4EB/vunit/tree/petterssonmagnus-xsim or would you like to rebase it first?

imd1, 2019-10-24:
I'm happy to make some progress on this, but I'm not too GitHub literate, so I might need some guidance on topics such as forking and rebase in this case. Should I fork 1138-4EB/vunit and work on the xsim branch? How should I rebase?

1138-4EB, 2019-10-24:
I suggest to clone this repo first:

```sh
git clone --recurse-submodules https://github.com/VUnit/vunit
cd vunit
```

Then, add my fork as a remote:

```sh
git remote add 1138-4EB https://github.com/1138-4EB/vunit
git fetch 1138-4EB
```

Now, checkout the branch:

```sh
git checkout -b xsim 1138-4EB/petterssonmagnus-xsim
```

Last, you can either fix this version, or rebase:

```sh
git rebase origin/master
```

Rebasing will require you to fix possible conflicts between existing modifications in branch xsim and recent additions to master. Likely in Pythons sources, since no VHDL sources are modified in branch xsim: https://github.com/VUnit/vunit/compare/master...1138-4EB:petterssonmagnus-xsim

If you don't know how to fix conflicts, just `git rebase --abort` and you will be back in `xsim` (`1138-4EB/petterssonmagnus-xsim`).

imd1, 2019-10-31:
I've applied the suggested patch to string_ptr_pkg-body-2002p.vhd and now I get an error in the file that is compiled immediately after...

```
ERROR: [VRFC 10-3782] `string_ptr_pkg` is not compiled into `vunit_lib` [/usr/lib/..../vhdl/logging/src/ansi_pkg.vhd:10]
```
The console output shows passes for string_ptr_pkg-body-2002p.vhs and string_ptr_pkg.vhd immediately before a fail for ansi_pkg.vhd

Any ideas?

1138-4EB, 2019-10-31:
@imd1, on the one hand, it seems that you are trying to compile sources with `vhdl_standard=08`, which is likely to fail because VHDL 2008 support in Vivado is quite limited. I suggest to use `vhdl_standard=93`. On the other hand, bear https://github.com/VUnit/vunit/issues/209#issuecomment-335524828 in mind.

haggaie, 2019-11-12:
I have a question related to Vivado & Xsim. I'm trying to use a testbench that uses Vivado's AXI Verification IP libraries. I added the IP similarly to the example in the `examples/vhdl/vivado` dir (skipping the compilation of the Vivado library because this is only for 3rd party simulators as far as I can see). Now I'm getting errors when building the XCI's generated SystemVerilog file saying that:
```
ERROR: [VRFC 10-91] axi_vip_pkg is not declared
```
The `axi_vip_pkg.sv` file is part of the Vivado library (found at `/opt/Xilinx/Vivado/2018.2/data/xilinx_vip/hdl/`), and I thought it should find it automatically when using xsim.

Do you think this is an issue with how xsim is run? Or should I add this file manually to the project?

haggaie, 2019-11-20:
> ```
> ERROR: [VRFC 10-91] axi_vip_pkg is not declared
> ```
This was solved by adding `-L xilinx_vip` to the compiler command line.

qarlosalberto, 2020-01-15:
I would like help with Xsim support.

@eine I am really interested in common VUnit functionalities and array library. My goal is to simulate a test as this https://github.com/qarlosalberto/cordicHDL/blob/develop/tb/cordic_top_tb.vhd with Xsim
What could I need?

eine, 2020-01-15:
@qarlosalberto, see https://github.com/VUnit/vunit/issues/209#issuecomment-545942699. I believe that the codebase didn't change significantly in that regard during the last couple of months.

qarlosalberto, 2020-01-16:
I'm working on with the setup: 
- Vivado 2019.2
- 1138-4EB/petterssonmagnus-xsim
- VUNIT_VHDL_STANDARD=93

I have applied the change:

```
    impure function length (
      ref : natural
    ) return integer is
    variable s : string_access_t := ptrs(ref);
    begin
      return s'length;
    end;
```

I have applied some changes and now I have this error: 

```
ERROR: [VRFC 10-3763] type error near 'arr.data' ; expected type 'natural' [/home/carlos/repo/vunit/vunit/vhdl/data_types/src/integer_array_pkg-body.vhd:157]
ERROR: [VRFC 10-3340] formal 'ref' has no actual or default value [/home/carlos/repo/vunit/vunit/vhdl/data_types/src/integer_array_pkg-body.vhd:157]
ERROR: [VRFC 10-3464] indexed name prefix type 'integer' is not an array type [/home/carlos/repo/vunit/vunit/vhdl/data_types/src/integer_array_pkg-body.vhd:157]
```

qarlosalberto, 2020-01-16:
My fault, I have been a wrong change. But I have a error with:

```
  impure function length (
    ptr : ptr_t
  ) return integer is begin
    return ptrs(ptr.ref)'length;
  end;
```

eine, 2020-01-16:
See https://github.com/VUnit/vunit/issues/209#issuecomment-545942699.

qarlosalberto, 2020-01-16:
It is correct the change? 

```
--  impure function length (
--    ptr : ptr_t
--  ) return integer is begin
--    return ptrs(ptr.ref)'length;
--  end;
 
  impure function length (
    ptr : ptr_t
  ) return integer is
    variable s : string_access_t := ptrs(ptr_t.ref);
  begin
    return s'length;
  end;
```

eine, 2020-01-16:
Worth a try.

qarlosalberto, 2020-01-16:
I have compiled all libraries :)

But I don't know why Vivado say that they aren't compiled in library vunit_lib...
```
=== Command output: ===
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit_vivado_example/mux_tb.vhd" into library work
ERROR: [VRFC 10-2987] 'string_ops' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:16]
ERROR: [VRFC 10-2987] 'dictionary' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:17]
ERROR: [VRFC 10-2987] 'path' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:18]
ERROR: [VRFC 10-2987] 'print_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:19]
ERROR: [VRFC 10-2987] 'log_levels_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:20]
ERROR: [VRFC 10-2987] 'logger_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:21]
ERROR: [VRFC 10-2987] 'log_handler_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:22]
ERROR: [VRFC 10-2987] 'log_deprecated_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:23]
ERROR: [VRFC 10-2987] 'ansi_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:24]
```

qarlosalberto, 2020-01-16:
With this line I compile VUnit libraries into vunit_lib, but I have the same error:

https://github.com/eine/vunit/blob/petterssonmagnus-xsim/vunit/xsim_interface.py#L103

```
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit/vunit/vhdl/run/src/run_deprecated_pkg.vhd" into library vunit_lib
Compiling into vunit_lib: ../vunit/vunit/vhdl/check/src/check_api.vhd                           passed
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit/vunit/vhdl/check/src/check_api.vhd" into library vunit_lib
Compiling into vunit_lib: ../vunit/vunit/vhdl/check/src/check.vhd                               passed
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit/vunit/vhdl/check/src/check.vhd" into library vunit_lib
Compiling into vunit_lib: ../vunit/vunit/vhdl/check/src/check_deprecated_pkg.vhd                passed
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit/vunit/vhdl/check/src/check_deprecated_pkg.vhd" into library vunit_lib
```

```
=== Command output: ===
INFO: [VRFC 10-163] Analyzing VHDL file "/home/carlos/repo/vunit_vivado_example/mux_tb.vhd" into library tb_lib
ERROR: [VRFC 10-2987] 'string_ops' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:15]
ERROR: [VRFC 10-2987] 'dictionary' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:16]
ERROR: [VRFC 10-2987] 'path' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:17]
ERROR: [VRFC 10-2987] 'print_pkg' is not compiled in library 'vunit_lib' [/home/carlos/repo/vunit_vivado_example/mux_tb.vhd:18]
```

qarlosalberto, 2020-01-17:
@imd1 Did you get it to work?

qarlosalberto, 2020-02-17:
To use XSim with VUnit VHDL libraries isn't possible. XSim doesn't support variable access types, for example:

acc := new vav_t'(0 to acc'length + 2**16 => null);

This is a big problem, since it is used in the basic types of VUnit. 

I'm woking in a very basic VUnit funcionalities in this repo:

https://github.com/softwareradiosystems/vunit

This is an example of VUnit with VHDL Xsim:

https://github.com/softwareradiosystems/vunit/tree/xsim-rebase-v2/examples/vhdl/xsim-test

Piressss, 2020-03-23:
@qarlosalberto Do you have any news?

qarlosalberto, 2020-03-23:
I'm using VUnit with Xsim in my job. It's a basic support:

https://github.com/softwareradiosystems/vunit/tree/xsim-rebase-v2

Although you can't use the VUnit libraries (only array). What do you want to do? I can say you if Xsim branch supports it.

Piressss, 2020-03-23:
We use the ModelSim integrated with Vunit, but we have a limit of 3 licenses for that job. I would like to test the VSim integrated with Vunit for small and simples testbench, using the Vunit Check Library.

qarlosalberto, 2020-03-23:
Here you can see the implemented functions of Check Library for Xsim:

https://github.com/softwareradiosystems/vunit/tree/xsim-rebase-v2/vunit/vhdl/xsim/check/src

You could add more functions.

Piressss, 2020-03-23:
I'll try to use it. Thanks.

eine, 2020-03-23:
> We use the ModelSim integrated with Vunit, but we have a limit of 3 licenses for that job. I would like to test the VSim integrated with Vunit for small and simples testbench, using the Vunit Check Library.

Have you considered to use GHDL for this purpose instead of XSim? Since it is free and open source software, you don't have any license-related constraint; and it is very lightweight compared to Vivado.

qarlosalberto, 2020-03-23:
@eine  is right, if it is possible use GHDL. A simulation in Xsim can be 20 times slower than in GHDL...

Piressss, 2020-03-23:
I'm not sure if the GHDL is a homologated tool by the IT team here.

Let me ask you a question. I created the environment variable VUNIT_XSIM_PATH and VUNIT_SIMULATOR = xsim.

When I try to compile, I get the following error:
```

  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\ui\__init__.py", line 683, in main
    all_ok = self._main(post_run)
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\ui\__init__.py", line 727, in _main
    return self._main_compile_only()
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\ui\__init__.py", line 865, in _main_compile_only
    simulator_if = self._create_simulator_if()
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\ui\__init__.py", line 749, in _create_simulator_if
    args=self._args, output_path=self._simulator_output_path
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\sim_if\xsim_interface.py", line 47, in from_args
    return cls(prefix=prefix, output_path=output_path, gui=args.gui)
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\sim_if\xsim_interface.py", line 67, in __init__
    self._xvlog = self.check_tool('xvlog')
  File "C:/Users/fepires/Downloads/vunit-xsim-rebase-v2\vunit\sim_if\xsim_interface.py", line 61, in check_tool
    raise Exception('Cannot find %s' % tool_name)
Exception: Cannot find xvlog
```

Have you already see it before?

eine, 2020-03-23:
> I'm not sure if the GHDL is a homologated tool by the IT team here.

Actually, GHDL can be extracted/built to a user directory, so it can be used without admin permissions. Anyway, you might want to ask the IT team about it. GHDL is available in several distribution repositories, and they might consider to add it. As a matter of fact, GHDL and VUnit are used as a complement to vendor tools in many companies/institutions without managers being aware. I'm NOT enforcing you to go against any rule, but just letting you know that adoption of VUnit and GHDL is done bottom-to-top in many organisations.

> Let me ask you a question. I created the environment variable VUNIT_XSIM_PATH and VUNIT_SIMULATOR = xsim.

Which is the content of VUNIT_XSIM_PATH? Is it pointing to a directory or a binary?

Piressss, 2020-03-23:
> Actually, GHDL can be extracted/built to a user directory, so it can be used without admin permissions. Anyway, you might want to ask the IT team about it. GHDL is available in several distribution [...]

I found the problem, is the VUNIT_XSIM_PATH, I typed wrong path.

Piressss, 2020-03-23:
Another question.

I got the following error:

`ERROR: [XSIM 43-4316] Can not find file: E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\sync_lib`

The path library was not created, but I can see a directory called 'xsim.dir' at the same level of 'vunit_out':

```
    Directory: E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\xsim.dir

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        3/23/2020   1:33 PM                sync_lib
```

Is it correct?

qarlosalberto, 2020-03-23:
Could you share your run.py?

Piressss, 2020-03-23:
```
#!/usr/bin/env python
import sys, os, subprocess, shutil
import os.path as p
#from vunit import VUnit
from data_checker import ZeroSamplesChecker

sys.path.append('C:/Users/fepires/Downloads/vunit-xsim-rebase-v2/')

from vunit import VUnit, VUnitCLI
from vunit.sim_if.factory import SIMULATOR_FACTORY

def main():
    root_dir = p.join(p.dirname(__file__), '..', '..', '..', '..', '..')
    #root_dir = root_dir.replace("\\", "/")
    lib_path = p.join(p.dirname(__file__), '..')
    lib_path = lib_path.replace("\\", "/")

    vhdl_standard = '2008'

    # Check Vsim 
    simulator_class = SIMULATOR_FACTORY.select_simulator()
    simname = simulator_class.name
    print(simname)

    # Verify the version of the Modelsim
    mver = os.environ['MSIM_VER']
    if not os.path.isfile('msim_ver.log'):
        ofile=open('msim_ver.log','w')
        ofile.write(mver)
        ofile.close()
        if os.path.exists('vunit_out/'):
            shutil.rmtree('./vunit_out')
    else:
        ifile=open('msim_ver.log','r')
        tmp = ifile.readline()
        ifile.close()
        if mver != tmp:
            print("ModelSim version changed since the last run")
            ofile=open('msim_ver.log','w')
            ofile.write(mver)
            ofile.close()
            if os.path.exists('vunit_out/'):
                shutil.rmtree('./vunit_out')

    # Path da XILINX
    xilinx_path = os.environ['XILINX_VIVADO']
    xilinx_path = xilinx_path.replace("\\", "/")

    ui = VUnit.from_argv()
    ui.add_com()

    # Adding testbenchs. Its name must end with '_tb'
    lib = ui.add_library("axi_stream_lib_tb", vhdl_standard)
    # TB
    lib.add_source_file(p.join(lib_path, 'simul', "axi_length_adjust_tb.vhd"))
    add_axi_length_adjust_test(lib.entity("axi_length_adjust_tb"))

    lib.add_source_file("axi_frame_repeater_tb.vhd")
    frame_repeater_test(lib.entity("axi_frame_repeater_tb"))

    lib.add_source_file("axis_accum_tb.vhd")
    add_axis_accum_test(lib.entity("axis_accum_tb"))

    lib.add_source_file("axi_truncate_tb.vhd")
    truncate_test(lib.entity("axi_truncate_tb"))

    lib.add_source_file("axi_stream_insert_tlast_tb.vhd")
    add_axi_stream_insert_tlast_test(lib.entity("axi_stream_insert_tlast_tb"))
    lib.add_source_file("axis_sync_fifo_tb.vhd")
    add_axis_sync_fifo_test(lib.entity("axis_sync_fifo_tb"))

    lib.add_source_file("axis_gen_pckg_tb.vhd")
    add_axis_gen_pckg_test(lib.entity("axis_gen_pckg_tb"))
    lib.add_source_file("axi_frame_decimation_tb.vhd")
    add_axi_frame_decimation_test(lib.entity("axi_frame_decimation_tb"))

    lib.add_source_file("axis_zero_samples_tb.vhd")
    add_axis_zero_samples_test(lib.entity("axis_zero_samples_tb"))

    lib.add_source_file(xilinx_path + "/data/verilog/src/glbl.v")

    # IP
    if not os.path.isdir('axi_stream_lib'):
        os.system("vivado -mode tcl -source vivado_generate_ip.tcl")
        file_name = os.listdir("./")
        for file_i in file_name:
            if file_i.endswith('.jou') or file_i.endswith('.jou'):
                os.remove(file_i)

    # Libs
    lib = ui.add_library("osvvm_lib")
    lib.add_source_files(p.join(root_dir, 'libs', 'osvvm_lib', 'tags', '1.0.0', 'code', '*.vhd'))

    lib = ui.add_library("base_lib")
    lib.add_source_files(p.join(root_dir, 'libs', 'base_lib', 'tags', '1.1.0', 'code', '*.vhd'))

    lib = ui.add_library("sync_lib")
    lib.add_source_files(p.join(root_dir, 'libs', 'sync_lib', 'tags', '1.1.0', 'code', '*.vhd'))

    lib = ui.add_library("axi4_test")
    lib.add_source_files(p.join(root_dir, 'libs', 'axi4_test', 'tags', '1.0.0', 'code', '*.vhd'))

    lib = ui.add_library("check_lib")
    lib.add_source_files(p.join(root_dir, 'libs', 'check_lib', 'branches', 'trunk', 'code', '*.vhd'))

    ui.main()

qarlosalberto, 2020-03-23:
Could you share the complete log? With --verbose

Piressss, 2020-03-23:
> Could you share the complete log? With --verbose

```
Compiling into sync_lib:          ..\..\..\..\sync_lib\tags\1.1.0\code\syncr_vector.vhd                                                failed
=== Command used: ===
C:\XilinxSysEd\Vivado\2017.4\bin\xvhdl.bat E:\svn\m200\libs\sync_lib\tags\1.1.0\code\syncr_vector.vhd -2008 -work sync_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\sync_lib -L vunit_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\vunit_lib -L axi_stream_lib_tb=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\axi_stream_lib_tb -L osvvm_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\osvvm_lib -L base_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\base_lib -L sync_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\sync_lib -L axi4_test=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\axi4_test -L check_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\check_lib -L axi_stream_lib=E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\axi_stream_lib

=== Command output: ===
ERROR: [XSIM 43-4316] Can not find file: E:\svn\m200\libs\axi_stream_lib\branches\trunk\simul\vunit_out\xsim\libraries\sync_lib
```

qarlosalberto, 2020-03-23:
Could you have a previous error/warning with some file in the lib sync_lib and it isn't compiled? It is the folder sync_lib empty?

Piressss, 2020-03-23:
> Could you have a previous error/warning with some file in the lib sync_lib and it isn't compiled? Is it the folder sync_lib empty?

syncr_vector.vhd is the first code being compiled, and the folder sync_lib is empty.

[60 more replies]