VUnit/vunit #211: Human-readable vs hashed test folder

gsorrenti, 2016-11-23:
Human readable folder are better for manual exploration while hashed folder are self protected against special characters and long path and are better for automated environment.

joshrsmith, 2016-11-24:
At first I was hesitant with the move to hashed folder name, but it has grown on me. I especially like the fact that I don't need to worry about the test case names anymore. I can make them as long as I want, with whatever characters I want.  The text file is a good "decoder ring" to figure out which folder to look in for outputs. I don't think we have much documentation on the output/working folder structure, but if we did it would be good to point out how the structure is generated.

gsorrenti, 2016-12-01:
My suggestion was not to add a configuration but only an automated switch to hashed path only when really necessary. 
As an example should be to apply hash path only when path exceed maximum length. In those cases you can substitute the leading part of the path with an hash of this part and leave the trailing part unchanged...this way you can easily identify different configuration of the same test.
In case of special character the hashed path should be applied.
Of course a dictionary should be always created to be compliant with automations.

kraigher, 2017-07-19:
I added some documentation of the test output folder: http://vunit.github.io/cli.html#test-output-paths.
I also added human readable test output paths with some limitations on Windows and a few related environment variables if tuning is required.

closed 2017-07-19 by kraigher