VUnit/vunit #215: Active-HDL do not support sv2k12 option

gsorrenti, 2016-12-04:
Active-HDL do not support sv2k12 option while Riviera do. Now I workaround the problem removing this option from active-hdl interface.

kraigher, 2016-12-04:
What version of Active-HDL are you using? What is the output of vlog -help?

kraigher, 2016-12-04:
The solution to this issue is to make the verilog standard configurable just like the vhdl standard is.

gsorrenti, 2016-12-04:
Active-HDL 9.2

gsorrenti, 2016-12-04:
I think should be done two action:
1) add configurable verilog standard
2) if the option sv2k12 is not supported by active-HDL, Vunit should rise an error if it is configured to build sv2k12 standard with active-hdl

LarsAsplund, 2016-12-04:
We started our support of Active-HDL at version 9.3. Is it possible for you to update.

closed 2017-01-05 by kraigher