VUnit/vunit #217: VUnit not passing signals to the top level?

Brian-Drummond, 2016-12-06:
I refactored the VUnit tests into an entity as below.
Oddly if I comment out the lines labelled "Vunit additions" in stimulus_response, (to build it standalone in ghdl), VUnit runs both the toplevel (with signals passed up) and stimulus_response (no signals passed up)... but if I don't pass the generic to the sub-unit, neither runs...

Testcase...

[run.py.txt](https://github.com/VUnit/vunit/files/633882/run.py.txt)

[stimulus_response.vhd.txt](https://github.com/VUnit/vunit/files/633899/stimulus_response.vhd.txt)
[tb.vhd.txt](https://github.com/VUnit/vunit/files/633900/tb.vhd.txt)

LarsAsplund, 2016-12-06:
The VUnit scanner identifies testbenches as entities with a `runner_cfg` generic. In your case there are two such files so VUnit thinks you have one testbench without any test cases (which is ok) and one testbench with one test case. `python run.py --list` will give

``` console
lib.stimulus_response.OSVVM_tests
lib.tb.all
Listed 2 tests
```
If you change the name of `runner_cfg` in `stimulus_response.vhd` to something else like `nested_runner_cfg` you will get this when running the same command

``` console
lib.tb.all
Listed 1 tests
```
VUnit assumes that your test cases are defined in the same file as the `runner_cfg` generic. If not  you have to add the following to `run.py`

``` python
tb = lib.entity("tb");
tb.scan_tests_from_file("stimulus_response.vhd")
```

Run the command again and you will have what you want

``` console
lib.tb.OSVVM_tests
Listed 1 tests
```

Note that the `Test_Random_Valid_Transactions` procedure ends with a `wait` such that the simulation will stop with nothing more to do. A stop without passing `test_runner_cleanup` is considered an error even though there were no failing asserts. The reason is to detect errors like division by zero which cause simulations stops despite that there was no failing asserts.

kraigher, 2016-12-06:
Regarding scan_tests_from_file read the vunit documentation: http://vunit.github.io/python_interface.html#vunit.ui.TestBench.scan_tests_from_file
The function is added to support legacy setups where the main test process is not in the top level entity for simulation which is an assumption in VUnit to have dependency scanning without needing to elaborate the design in Python. I claim that any test bench architecture can be refactored to have the main process on the top level test bench without loss of functionality or ease of maintainance.

closed 2016-12-06 by LarsAsplund