VUnit/vunit #220: GHDL Error in the Check Library with v1.3.0

nhajidin, 2017-01-16:
When running the UART example, I get the following error:

```
Compiling C:\Python27\lib\site-packages\vunit\vhdl\check\src\check.vhd into vunit_lib ...
are_trees_equal: cannot handle IIR_KIND_FUNCTION_CALL (C:\Python27\lib\site-packages\vunit\vhdl\check\src\check.vhd:335:55)

******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2014 (20140331)
In directory: D:\\vunit\examples\vhdl\uart\
Command line:
D:\ghdl-0.33\bin\ghdl.exe -a --workdir=D:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\vunit_lib --work=vunit_lib --std=08 -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\vunit_lib -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\osvvm -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\uart_lib -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\tb_uart_lib C:\Python27\lib\site-packages\vunit\vhdl\check\src\check.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:63
******************************************************************
Failed to compile C:\Python27\lib\site-packages\vunit\vhdl\check\src\check.vhd with command:
D:\ghdl-0.33\bin\ghdl -a --workdir=D:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\vunit_lib --work=vunit_lib --std=08 -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\vunit_lib -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\osvvm -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\uart_lib -PD:\\vunit\examples\vhdl\uart\vunit_out\ghdl\libraries\tb_uart_lib C:\Python27\lib\site-packages\vunit\vhdl\check\src\check.vhd
```

This is using GHDL 0.33 and vuint 1.3.0 on Windows10 with Python 2.7.13. It works fine using vunit 1.2.0.

LarsAsplund, 2017-01-16:
@nhajidin I managed to recreate your problem and it seems to be a bug in GHDL 0.33. I think I have a workaround that I will test more thoroughly tomorrow. Until then I recommend that you revert to 1.2.0 or upgrade to the development version of GHDL (0.34).

fordred, 2017-01-17:
I can confirm a similar error in the [Fedora 25 repository package (0.34dev-0.20160702git50d0507.0.fc25)](https://apps.fedoraproject.org/packages/ghdl).

My workaround was to build the latest version of GHDL from the source available on [GitHub](https://github.com/tgingold/ghdl).

LarsAsplund, 2017-01-17:
@nhajidin @fordred I've created a new release 1.3.1 to address this. Thanks for reporting.

closed 2017-01-17 by LarsAsplund