VUnit/vunit #223: Modelsim startup.do

drizzd, 2017-01-27:
I would like to do something like `vsim -gui -do "source startup.do" lib.design`, where startup.do will typically do stuff like adding signals to the waveform viewer. Is this possible with VUnit?

I am trying to use `VUnit.set_sim_option("modelsim.vsim_flags.gui", ["-do", "\"source startup.do\""])`, but I get a warning `Ignoring nested -do argument: "source startup.do"` and the script is not executed. I think the reason is that VUnit does `vsim -gui -do "do gui_load.do"`, and gui_load.do in turn calls vsim. Together that amounts to `vsim -gui -do "vsim -do \"source_startup.do\" lib.design"`, i.e. a "nested do"

I guess I could teach VUnit to run `vsim -gui -do "source startup.do" lib.design` with all the appropriate options, but then I cannot run vunit_run in the GUI any more, right?

kraigher, 2017-01-27:
We could add a new simulation option such as "modelsim.startup_file.gui" that allows specifying a file that will be sourced after loading the top level in vunit_load.

We could also add a comment pragma to allow setting such a file from within a test bench to avoid having to go through run.py.

This would cover the use case that you want to have a waveform configuration script always run when opening a test bench in gui mode.

drizzd, 2017-01-28:
Sounds good to me. I don't think I would be using the comment pragma, though.

kraigher, 2017-01-28:
You can check out the issue_223 branch and see if the feature is to your liking.

kraigher, 2017-01-28:
I also added a TCL prodecure exposed in the GUI named `vunit_startup` such that you can re-run your custom startup file in the GUI. Might be useful when developing the startup file and adding signals to it etc.

kraigher, 2017-01-28:
I do not now if "startup_file" is the best name for this. Anyone can think of something better? Otherwise I might let it stick.

joshrsmith, 2017-01-28:
Some other possibilities: init_script, custom_init_script

kraigher, 2017-01-29:
I decided to call it `init_file`. It is on master now.

drizzd, 2017-01-30:
Tested here and works like a charm. It is very reassuring that even feature requests are implemented so quickly. Thank you!

Here is a possibly helpful observation for future users. The working directory in init_file is different from the command line working directory. In order to run scripts which reside in the directory that also contains init_file, resolve the path like this:

    # Call wave.do from init_file.do
    do [file join [file dirname [info script]] wave.do]

closed 2017-01-30 by drizzd