VUnit/vunit #225: Add code coverage support for Aldec Riviera-Pro

noasic, 2017-02-07:
Code coverage is currently supported only for ModelSim. I'd like to have it supported in Riviera-Pro too.

bbmag-bard, 2017-02-09:
That would be great! I'm also using the Riviera-Pro simulator and would be looking forward to using this feature.

noasic, 2017-05-12:
I've added experimental support to Riviera-Pro code coverage on a local branch but I don't have permission to push that branch to the Github repository:

> remote: Permission to VUnit/vunit.git denied to noasic.
> fatal: unable to access 'https://github.com/VUnit/vunit.git/': The requested URL returned error: 403

How am I supposed to proceed?

kraigher, 2017-05-12:
@noasic The process is: You do a "pull request" in the GitHub web GUI from your fork to the VUnit repo. Then you pass all automated tests and fix our comments on your commits and then we press the merge button.

commit 358493ec (2017-05-21): Adds code coverage support for RivieraPRO. Fixes #225

Also makes code coverage flag trigger recompile of source files due different compile options for coverage.
closed 2017-05-21 by kraigher