VUnit/vunit #226: fix missing --gui=False option for ghdl backend

StefanD986, 2017-02-09:
When using the ghdl backend, it does not seem possible to generate .ghw/.vcd waveform files without automatically opening gtkwave: I tried the --gui=False switch that is listed in the documentation (https://vunit.github.io/cli.html#opening-a-test-case-in-simulator-gui section _ghdl_), but this is not accepted by the `run.py` script.

kraigher, 2017-02-09:
GHDL has no gui, thus the --gui flag is not available. There is an --gtkwave flag though to open the result in gtkwave after simulation.

kraigher, 2017-02-09:
So you are missing a way to emit waveform files without also opening a GUI? Could you clarify your use case? Do you want configure this permanently in the run.py file or set it from time to time on the command line when doing manual debugging?

StefanD986, 2017-02-09:
I know that ghdl has no GUI, but I was assuming that the "--gui" flag would refer to the gtkwave gui for the ghdl simulation backend.

Maybe I should rename the issue to make it clearer: There should be a way to produce the waveform data without automatically opening gtkwave.

kraigher, 2017-02-11:
I have added what you want now. --gui replaces old --gtkwave flag.
Use --gtkwave-fmt to explicitly set format when using --gui.
Use --gtkwave-fmt without --gui to save files without launching gtkwave.

closed 2017-02-11 by kraigher

StefanD986, 2017-02-11:
Thank you! This exactly what I needed.

A bit more about the use case: I wanted the option to generate the waveform files without automatically opening gtkwave because it is always quite a bit of work to add the signals of interest to the gtkwave waveform view. What I usually do is: I simply keep my gtkwave session open between simulation runs, and then reload the file. This way I do not need to add my signals of interest again in gtkwave, it just updates the waveform graph view with the new data.