VUnit/vunit #227: add_config ordering

drizzd, 2017-02-10:
It would be nice if test configurations were executed in the same order that they are added with add_config. I think this could be achieved by replacing self._configs = {} with self._configs = [collections.OrderedDict](https://docs.python.org/2/library/collections.html#collections.OrderedDict). The ordering of a regular [dictionary](https://docs.python.org/2/tutorial/datastructures.html#dictionaries) is undefined: "It is best to think of a dictionary as an unordered set of key: value pairs"

kraigher, 2017-02-10:
Currenty the order is lexicographical, why is this better? Is there any objective reason?
Maybe we should have a poll if it is subjective. The same principle can be applied to the order entities are added. Now tests are run in lexicographical order based on library name and entity name.

drizzd, 2017-02-11:
Since the test configurations are added sequentially, I think it is intuitive to execute in the same order. This way, the user can easily choose the ordering.

My immediate motivation is to run one of the test configurations first, because it is faster than the others and catches most of the issues. I can achieve this by prefixing the configuration name with "AAA_", since lexicographical ordering is enforced. It's just a bit ugly.

On the other hand, changing the ordering would break the expectation of existing users. So I can live with the lexicographical ordering. Maybe we could also add it to the documentation of Testbench.add_config.

kraigher, 2017-02-22:
This is now implemented in v2.0.0

closed 2017-02-22 by kraigher