VUnit/vunit #228: modelsim vunit_restart doesn't work

nathanaelhuffman, 2017-02-10:
Output with no source file changes:
``` 
vunit_restart
# Re-compiling using command <my_path2_python>\python.exe -u run.py --compile lib.tb_xxx.yyy --gui
# Re-compile failed
```
Output with a source file change:
```
vunit_restart
vunit_restart
# Re-compiling using command <my_path2_python>\python.exe -u run.py --compile lib.tb_xxx.yyy --gui
# Compiling tb\tb_xxx.vhd into lib ...
# ** Warning: (vcom-14) Failed to open "modelsim.ini" specified by the MODELSIM environment variable.
# 
# The system cannot find the file specified. (GetLastError() = 2)
# Re-compile failed
```

modelsim 10.4a if that matters.

kraigher, 2017-02-10:
Have you manually set the MODELSIM environment variable?

nathanaelhuffman, 2017-02-10:
No.  I have these set, but no env variable named modelsim:
[image]

Is a modelsim env required?  If so where is this documented?

kraigher, 2017-02-10:
No it is not required. It might interfere if set though.

kraigher, 2017-02-10:
Sure it is not set? Have you run `env`?

kraigher, 2017-02-10:
Anyway I pushed a commit which protects against MODELSIM environment variable being set.

nathanaelhuffman, 2017-02-10:
In my run.py:
```
print('\n'.join(sorted(os.environ.keys())))
vu.main()
```
Here's the relevant output:
[image]

So no modelsim passing in to the python env when I do a run.py in the command shell.

When I run the vunit_restart in modelsim, it prints the same stuff again in the modelsim terminal, but this time, there are modelsim definitions.  This is even with your latest commit:
[image]

```
>python run.py --version
1.4.1-pre
```

kraigher, 2017-02-10:
yes vsim seems to set environment variables when started, that was the problem since VUnit is invoked again from vsim. With the new commit that should not matter since MODELSIM is removed in the invoked VUnit process again

kraigher, 2017-02-10:
But currently I just remove MODELSIM, there migth be other variabler that should be removed as well

nathanaelhuffman, 2017-02-10:
Ok I'll try to compare pre-run and post-run and see what things I can figure out.  
It seems that I should be able to 
I see the MODEL_TECH variables are added as well.  This is 10.4a SE, not sure how common that is out there but maybe there are some other things going on there.

Are there options for more verbosity on the vunit_restart to see what kind of failure is occurring once the executable is called?  The restart with no changes which doesn't actually try to recompile also returns with error.

nathanaelhuffman, 2017-02-10:
Ok so I have a warning below, that isn't fatal to the simulation, running normally with vunit everything works, but I think tcl's catch is detecting this as a failure.
[image]

I hacked the vunit source to print the return code in main() and it is 0, but a few warnings are thrown like above.  It's been a while since I've had to deal with tcl so I have to do some reading to further debug what is going on.

I did change the vunit_compile command to use the 2nd variable in the catch statement and printed that too, called extra_info:
```tcl
if {[catch {close $chan} error_msg extra_info]} {
```
Output:
```
# -code 1 -level 0 -errorcode NONE -errorinfo {WARNING - C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd: failed to find a primary design unit 'all' in library 'altera_mf'
```

Thoughts?

nathanaelhuffman, 2017-02-11:
Ok, I think I understand what is going on here:

1. warnings in vunit get printed to stderr, even if the vunit process returns 0.  That's probably ok.
2. tcl command pipelines "generate errors exactly as described for exec" according to [here](http://wiki.tcl.tk/1241) but unlike exec, there's no --ignorestderr option so if anything is printed to stderr, the close of the pipeline will return with an error code which is what's going on here.

I also found a possible solution:
I copied the vunit_compile procedure into my own tcl file and changed the python subprocess.call() method to put stderr=sys.stdout instead of sys.stderr.

After making that change, the compilation is successful and restart works as expected.  Introducing a syntax error to a file will still make the subprocess call return non-zero and the compile is properly detected by tcl as a failure.

PR #229 (2017-02-11): TCL's command pipe will always return failing error codes on anything that prints to stderr
 Merge stderr to stdout from python so that TCL's catch will rely on the error code returned rather than being tricked by warnings printed to stderr.  Fixes #228
closed 2017-02-11 by kraigher