VUnit/vunit #231: Add Questa Support

cmon1, 2017-02-21:
I tried to get Questa 10.5 running and tried both the 32 Bit and the 64 Bit Version. As it turns out Mentor changed the encoding of modelsim.ini to UTF-8 and that yields following error when using it with VUnit:

> Traceback (most recent call last):
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\ui.py", line 689, in main
>     all_ok = self._main()
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\ui.py", line 723, in _main
>     self._compile(simulator_if)
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\ui.py", line 827, in _compile
>     continue_on_error=self._keep_compiling)
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\simulator_interface.py", line 119, in compile_project
>     self.setup_library_mapping(project)
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\modelsim_interface.py", line 168, in setup_library_mapping
>     mapped_libraries = self._get_mapped_libraries()
>   File "C:\tools\Python35-32\lib\site-packages\vunit_hdl-0.71.0-py3.5.egg\vunit\modelsim_interface.py", line 248, in _get_mapped_libraries
>     cfg.read(self._modelsim_ini)
>   File "C:\tools\Python35-32\lib\configparser.py", line 696, in read
>     self._read(fp, filename)
>   File "C:\tools\Python35-32\lib\configparser.py", line 1012, in _read
>     for lineno, line in enumerate(fp, start=1):
>   File "C:\tools\Python35-32\lib\encodings\cp1252.py", line 23, in decode
>     return codecs.charmap_decode(input,self.errors,decoding_table)[0]
> UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 3239: character maps to <undefined>

For now I just converted Questas modelsim.ini from UTF to 8859 and it works fine. But of course it would be nice if VUnit would resolve those encoding issues.

kraigher, 2017-02-22:
This has been fixed in later releases. You should upgrade to at least v1.0.0.

kraigher, 2017-04-03:
I am assuming my recommendation fixed your problem and closing the issue.

closed 2017-04-03 by kraigher