VUnit/vunit #232: VUnit fails if execute nested projects with GHDL

tivaliy, 2017-03-06:
I have a problem in executing VUnit tests from nested project with GHDL.
Let's suppose that we have some VHDL project with the following structure:
```
.
├── lib1
│   ├── adder.vhd
│   └── tb_adder.vhd
├── lib2
│   ├── d_ff.vhd
│   └── tb_dff.vhd
└── run.py
```

Here is the content of `run.py` file:

```python
import glob
import os

from vunit import VUnit

prj_root = os.path.dirname(os.path.abspath(__file__))
pattern = "*.vhd"

ui = VUnit.from_argv()

for root, dirs, _ in os.walk(prj_root):
    # Skip searching in hidden directories
    dirs[:] = [d for d in dirs if not d.startswith('.')]
    if glob.glob(os.path.join(root, pattern)):
        lib_name = os.path.basename(os.path.normpath(root)) + '_lib'
        # Create standalone libraries from directories that contain *.vhd files
        user_lib = ui.add_library(lib_name)
        user_lib.add_source_files(os.path.join(root, "*.vhd"))

ui.main()
```
This code works great with ModelSim as external toolchain, but fails with GHDL:

```bash
WARNING - /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang.vhd: package 'lang' previously defined in /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock.vhd
WARNING - /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang.vhd: package body 'lang' previously defined in /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock.vhd
WARNING - /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock_special_types200x.vhd: package 'lang_mock_special_types_pkg' previously defined in /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock_special_types93.vhd
WARNING - /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock_special_types200x.vhd: package body 'lang_mock_special_types_pkg' previously defined in /tmp/tpks/vunit/lib/python2.7/site-packages/vunit/vhdl/vhdl/src/lang/lang_mock_special_types93.vhd
test_lib
Traceback (most recent call last):
  File "run.py", line 19, in <module>
    user_lib = ui.add_library(lib_name)
  File "/tmp/tpks/vunit/local/lib/python2.7/site-packages/vunit/ui.py", line 441, in add_library
    self._project.add_library(library_name, abspath(path), vhdl_standard)
  File "/tmp/tpks/vunit/local/lib/python2.7/site-packages/vunit/project.py", line 74, in add_library
    assert allow_replacement
AssertionError
```
Any ideas or workaround

tivaliy, 2017-03-06:
Sorry, this issue can be marked as **Wrong**. In my case I've created `vunit` virtual environment in current project directory. This leads to error, because it's also try to analyze that virtualenv `vunit` folder.

LarsAsplund, 2017-03-06:
Ok, I'm closing it.

closed 2017-03-06 by LarsAsplund