VUnit/vunit #233: incorrect compile order with UVVM

nathanaelhuffman, 2017-03-14:
I get a bunch of these warnings:
 failed to find a primary design unit 'td_cmd_queue_pkg' in library 'bitvis_vip_uart' etc for every vvc.  Something about the generic packages maybe?

even though the target dependent files are added to the correct library.  I have cases where the dependency order works out ok, but I also have some where it does not and the files that are not found in the correct order

working compile order (not used in testbench though) from run.py -f:
[image]

Non-working order, you can see the td_queue_pkg is way at the bottom from run.py -f:
[image]

To make things even stranger, a co-worker can build this same test-bench just fine (his paths are different than mine, but all the same files) so there seems to be some kind of non-determinism here.  I get the same results under python3.6 and 2.7.

Sigasi with the project set up exports the compile order as expected.

LarsAsplund, 2017-03-14:
We have a pending update for recognizing package instantiations. Without that some dependencies are missing. Meanwhile you can use the `add_dependency_on` method (http://vunit.github.io/python_interface.html) to manually create the missing dependency.

kraigher, 2017-07-20:
I added dependency scanning usage of a package instance now. To discriminate against nested package instances I use the indentation level as a heuristic. It should work well in practice, if it doesn't we will have to pursue more complicated solutions.

commit 7a01c245 (2017-07-20): Add dependency scanning of the use of an instantiated package. Closes #233
closed 2017-07-20 by kraigher