VUnit/vunit #235: Codec compile error with array of custom type

joshrsmith, 2017-03-17:
Compiler: Riviera 2016.06
VUnit: 1.4.0

When generating codecs for a record with an array of a custom type, a compiler error is issued.
Should this be supported by the codec generator? Is there a work-around?

With a types in a types package defined as follows:
```
   type table_entry_t is record
      value : unsigned(4 downto 0);
      cell  : unsigned(10 downto 0);
   end record table_entry_t;

   type table_t is array (integer range <>) of table_entry_t;

   type cfg_t is record
      fixed_value : unsigned(7 downto 0);
      dly     : unsigned(7 downto 0);
      table   : table_t(31 downto 0);
   end record cfg_t;
```

I get an error when generating a codecs package (some info elided):
```
Compiling vunit_out\codecs\common\wfg_tb_msg_codecs_pkg.vhd into common ...
COMP96 ERROR COMP96_0124: "Illegal prefix for attribute 'left." "<path>\wfg_tb_msg_codecs_pkg.vhd" 509 89
COMP96 Compile failure 1 Errors 0 Warnings  Analysis time :  94.0 [ms]
Failed to compile <path>\wfg_tb_msg_codecs_pkg.vhd with command:
E:\Programs\Riviera\2016.06-x86\bin\vcom -quiet -j <path>/rivierapro -dbg -2008 -work common <path>\wfg_tb_msg_codecs_pkg.vhd
```

Here is the code referenced
```
  function decode (
    constant code : string)
    return table_t is
    constant range_length : positive := encode(integer'left)'length;
    function ret_val_range (
      constant code : string)
      return table_t is
      constant range_left : integer := decode(code(code'left to code'left + range_length - 1));
      constant range_right : integer := decode(code(code'left + range_length to code'left + 2 * range_length - 1));
      constant is_ascending : boolean := decode(code(code'left + 2 * range_length to code'left + 2 *range_length));
      variable ret_val_ascending : table_t(range_left to range_right);
      variable ret_val_descending : table_t(range_left downto range_right);
    begin
      if is_ascending then
        return ret_val_ascending;
      else
        return ret_val_descending;
      end if;
    end function ret_val_range;
    constant array_of_correct_range : table_t := ret_val_range(code);
    variable ret_val : table_t(array_of_correct_range'range) := (others => table_entry_t'left);
    variable index : positive := code'left + 1 + 2 * range_length;
  begin
    for i in ret_val'range loop
      decode(code, index, ret_val(i));
    end loop;

    return ret_val;
  end function decode;
```

Line 509 in snippet above is:
```
    variable ret_val : table_t(array_of_correct_range'range) := (others => table_entry_t'left);
```

joshrsmith, 2017-03-17:
Work-around:
```
type table_t is array (31 downto 0) of table_entry_t;
```

Note the fixed size of the table_t type.

LarsAsplund, 2017-03-30:
I'm in the process of renewing my license so I can't really test this myself but have you tried to replace

``` python
# @TODO: Remove workaround when Aldec issue SPT73021 is solved
if simulator_is('rivierapro', 'activehdl'):
    init_value = " := (others => %s'left)" % self.subtype_indication
else:
    init_value = ''
```

in `codec_vhdl_array_type.py` with just

``` python
init_value = ''
```

This workaround was added to handle unconstrained arrays on non-custom types which did not work in 2015.06. It was fixed in 2015.10 but I'm not sure if I ever verified that. Could you also run the unit tests on `com`after that?

LarsAsplund, 2017-05-03:
I simply removed the workaround which means that some use cases doesn't work with 2015.06. If someone really needs that I will consider a version dependent workaround

closed 2017-05-03 by LarsAsplund