VUnit/vunit #238: testbensch parser

asicnet, 2017-03-30:
As example  :        if (run ( "test_passe" ) )  then 
is not working 
solution:  changed testbench.py line 234 
from
_RE_VHDL_TEST_CASE = re.compile(r'\s+run\("(.*?)"\)', re.IGNORECASE)
to 
_RE_VHDL_TEST_CASE = re.compile(r'\W+run\s*\(\s*"(.*?)"\s*\)', re.IGNORECASE)

It is for a not restrictive VHDL coding

Best regards
AsicNet

LarsAsplund, 2017-03-30:
I think we should support that use case but since we're always targeting expressions we should limit the regexp to only accept whitespaces and opening parentheses before `run`. `\W` could lead to false positives.

asicnet, 2017-03-31:
This will be nice

_RE_VHDL_TEST_CASE = re.compile(r'[\(\s]+run\s*\(\s*"(.*?)"\s*\)', re.IGNORECASE)

kraigher, 2017-04-03:
@asicnet You seem to have done some work already. Can you make a pull request also with associated test cases? The tests are located in https://github.com/VUnit/vunit/blob/master/vunit/test/unit/test_test_bench.py#L85 
You can add tests for the cases not handled today to prove your new regex works.

asicnet, 2017-04-04:
Sorry, but i am not so deep in the environment that i can do here something quickly 
maybe you mean
contents='''\
if run( "Test_1" )
if run ( "Test_2")
if (run("Test_3"))
''')
And how i can test this.

LarsAsplund, 2017-04-04:
@asicnet  I would at least test these cases
``` vhdl
if run("Test 1")
if run("Test 2") or run("Test 3")
if run  ("Test 4")
if run( "Test 5"  )
if ((run("Test 6")))
if my_protected_variable.run("foo")
```

and expect it to find Test 1-6 but not anything from the last line

commit 61699d6a (2017-04-05): Make VHDL test case parsing more permissive. Closes #238
closed 2017-04-05 by kraigher