VUnit/vunit #240: Starting a simulation with multiple tops (xilinx glbl)

aodling, 2017-04-05:
Hi!

I'm wondering if there's an option to specify an additional top level module to simulate?

I'm simulating a lot of xilinx primitives /IP:s in verilog that requires glbl to be loaded...

Regards,
Andreas

kraigher, 2017-04-05:
@aodling That would typically be solved by using the simulator specific setting found in *sim_options* such as "modelsim.vsim_flags" which allows you to pass additional flags to the command that loads the top level for simulation. I do not know which simulator you are using though, there are similar mechanisms for RivieraPro. I do not think we want to add an abstraction layer for this in VUnit.

aodling, 2017-04-06:
@kraigher Thanks, I didn't realise that that was the way to do it. Works like a charm. Thanks for your help!

closed 2017-04-06 by aodling

Dhananjay20, 2019-12-30:
> @kraigher Thanks, I didn't realise that that was the way to do it. Works like a charm. Thanks for your help!

@aodling Can you tell me how you add glbl as a top module in script