VUnit/vunit #241: Case sensitivity for Libraries

cjchin, 2017-05-16:
We are getting error along the lines of this: 

Exmple Situation: 
 -Compiler: ModelSim
 -Vhdl file uses library: Common 
 -Python Run file contains: CommonLib = ui.add_library("Common")
 -Resultant Error: Cannot find expanded name "common.Interface".

Current fix: 
 -Change Python Run file contains: CommonLib = ui.add_library("common")

Suggested Resolution: 
 -Update Class method for add_library to convert input argument string to lower case. 

Please approve and I will proceed with updating the code.

kraigher, 2017-05-16:
It depends, either we change the data model or the lookup method when searching from identifiers from VHDL. We also support SystemVerilog and there libraries are case sensitive I believe thus I do not think it is sound to change the data model by converting the names to lower case immediately.

cjchin, 2017-06-02:
It sounds tough. Do you think many things might break if the data model is updated? 

I would like to help but im not sure where i should be looking. Can you let me know around should i start to look regarding this issue?

kraigher, 2017-06-02:
This is something that I would need to fix myself, it requires to much inside knowledge to create an elegant solution to this. If we change the data model it will break Verilog code that have upper case libraries. I will fix this somehow, for now a reasonable work-around is to use lower-case letters in the run.py file.

cjchin, 2017-06-03:
I see. Well.. You can always write if there is any minor tasks that I can assist with. Or maybe until i learn enough of the framework to propose a solution. thank you

commit 381c0887 (2017-06-03): Adds support for referencing non-lower case library names. Fixes #241.

Also forbids defining libraries with case-insensitive name collision.
closed 2017-06-03 by kraigher