VUnit/vunit #247: Change VUnit_Lib to Vunit

cjchin, 2017-06-02:
Since when we include VUnit library it is already implied that it is a library. So it seems more appropriate that VUnit_Lib should just simply be VUnit. 

What do you guys think? 

Although this change is backward incompatible...

cjchin, 2017-06-02:
_pkg also seems to be of similar topic.

LarsAsplund, 2017-06-02:
VUnit actually became a reserved word in VHDL-2008 so that library can't be named just that. The _pkg naming convention is commonly but not universally used. We are not even entirely consistent with our packages. Changing the naming would indeed break backward compatibility and I don't think it's worth that. After all, most people just include the `vunit_context`

closed 2017-06-02 by LarsAsplund