VUnit/vunit #248: Check_Not_Unknown for entire output port

cjchin, 2017-06-02:
It is quite common to just check to all signals in an output port. Would it be possible to implement a new function to have just one function to check all out signals?

LarsAsplund, 2017-06-02:
@cjchin What do you mean by "entire output port"? `check_not_unknown` exists for both `std_logic` and `std_logic_vector`

cjchin, 2017-06-03:
Take for example: 

entity Serializer is
	generic(INPUT_WORD_WIDTH : natural := 8);
	port(iSys       : in  rSys_logic;
		  iFreq      : in  std_logic_vector(15 downto 0);
		  iStart     : in  std_logic;
		  iByte      : in  std_logic_vector(INPUT_WORD_WIDTH - 1 downto 0);
		  oClkSig    : out std_logic;
		  oClkEnable : out std_logic;
		  oDatSig    : out std_logic;
		  oDatEnable : out std_logic;
		  oEnable    : out std_logic;
		  oReady     : out std_logic);
end entity Serializer;

Is there any way to just say Check_Not_Unknown_All(), which would simply check that all output ports are not unknown. Without having to explicitly write check_not_unknown for each signal?

LarsAsplund, 2017-06-03:
No , it's not possible for such a subprogram to know the ports of the
calling entity.

cjchin, 2017-06-04:
So i suppose the best way would be to make all output signals into one record type and do check_not_unknown on that?

LarsAsplund, 2017-06-04:
Yes, that you can do but you have to write that overloaded subprogram yourself. The `check_not_unknown` we provide has no knowledge of the elements in your custom record type so it can't figure out what do to with it.

The upcoming VHDL-2017 standard will provide record introspection so maybe we can improve on that once that is supported by the simulator vendors.

closed 2017-06-04 by LarsAsplund

cjchin, 2017-06-04:
I see got it, interesting. Record introspection sounds like generics in C#.

svenka3, 2017-06-05:
Hello,
  2 additional thoughts on this:

1. I believe through VHPI/FLI this is easy to do - I can volunteer to try that if you like. The downside is that we need to ship a .so file (or something similar) for each major EDA tool. Not a big deal, but be aware please. If you confirm your interest I can explore.

2. An off-line TCL based approach, through which we can spit out a VHDL code with checks for all ports and compile (Will require 2 passes with VHDL compiler). I believe this is easier than VHPI (We have done that before for Go2UVM apps), but flow is bit more complicated.

Comments?

Regards
Srini

cjchin, 2017-06-05:
My general thoughts into this is that, since checking for unknowns for output port seems to be a check that should be done regardless of what source code. 

The best would be if there will be some way to simply enable this check as simple as flicking on a switch, maybe even something directly enabled from the run.py script. That would be best. 

Both proposed solution seems to be capable of doing this. Although I think maybe the VHPI/FLI method may be a better solution since it would allow the current VUnit infrastructure to interface with it. 

I probably do not know enough to say if development should be attempted. 

@LarsAsplund do you think that something like this would be beneficial to VUnit? Or should we wait for record introspection?

kraigher, 2017-06-05:
I do not think the value of this feature justifies adding the complexity of FLI and .so files to VUnit.

LarsAsplund, 2017-06-07:
@svenka3, @cjchin In addition to @kraigher comment. From the beginning we've avoided to rely on simulator features not available in the less expensive licenses. Moving to VHPI/FLI type of interfaces would violate that principle.

Code generation is more in line we what we currently do for some more advance functionality that needs to work around the limitations in VHDL. We also have support for adding custom preprocessors that will be applied to all source files. So it would be possible to write a preprocessor that adds this kind of check to the entity and the file would only be compiled once after the preprocessor runs.

I can help you get started if you want to try that out. Just join our gitter room and you'll also get feedback from the general community. If it turns out to be a well-functioning concept we can consider including it into the VUnit core. If it turns out that there are "issues" preventing general applicability it would be better to publish it as an example but keep it separate from the VUnit core since "issues" tend to increase the support burden on our side. This is perfectly fine. Preprocessors are plugins and doesn't require integration with the core. It would actually be great to see people create an ecosystem of plugins and use examples that we could link from our website.

kraigher, 2017-06-07:
A draw back with preprocessors is that there is no incremental preprocessing support and that every file is unconditionally preprocessed each time you run the run.py file. This might be a performance problem for large project. Also the drawback is that the simulator will not see the code that is in you version control but rather a modified file in a temporary folder inside VUnit out.

LarsAsplund, 2017-06-07:
Waiting for VHDL-2017 to be implemented by your vendor is another approach. History tells us that can be a long wait.

There are of course other options available. Write a standalone code generator and keep the generated code under version control. Write a lint tool that fails your test suite when there are missing checks. If we just had a suitable open source VHDL parser for the task...

cjchin, 2017-06-08:
The idea of plugins seem to be very appealing(at least to me) and i believe that will be the key to the success of VUnit (although in retrospect I think VUnit is already very successful). Just like how VIM is hugely complemented by the whole plugin ecosystem and the community that exists around using these plugins (given how old VIM is and how widely used that it still is) 

A generalized way to allow for plugins will hugely benefit the community and generate new creative and innovative ideas that could be share and improved by everyone. Performance issue in terms of time required for each run is a problem but I do not believe it is a show stopper. Technology is always getting better and faster, and in this case, the potential capabilities vs status quo of what other testing frameworks can do severely outweighs this con. 

Have you ever thought about including VHDL parser as a possible direction for VUnit to grow along? Just like how Resharper of Jetbrains developed their own compiler to interpret C# for their refactoring tool analysis. Having VHDL parser integrated into the core of VUnit could open up so many opportunities. Linting, refactoring, code analysis. 

As FPGA technology continues to grow in the market, as AI technology scales, as Intel integration with FPGA technology propagate to mainstream. There exists a huge void for the tools needed for the job. 

At the same time, given that some of the initiatives of VHDL standard has been handed to the open source community..

LarsAsplund, 2017-06-08:
There is an ongoing discussion on VHDL Parsing in issue #174. I see the general parser as a separate project which VUnit may take advantage of to solve its own parsing needs. There is a trade-off between speed and completeness of the parsing. There are some initiatives out there which are referenced in the issue.

cjchin, 2017-06-08:
Interesting, do you see that pyVHDLParser as something that will gain traction?

LarsAsplund, 2017-06-08:
Not sure, it's something that just got started. I just hope that we eventually will have something that gain traction, this or some other initiative. I do think it will take time though. VHDL parsing is hard.