VUnit/vunit #249: Proposition for an additional constant parameter in check_stabie function

cjchin, 2017-06-04:
The current behaviour for check_stable is that : "If a second window is started before the previous is completed the second start event will be ignored and the window will be completed by the next end event."

Would it be good to consider an addition constant called _Allow_StartEvent_To_Restart_ where if this is set to true, then upon the assertion of a new start_event, to reset the stability check?

There are some cases where I find it inconvenient to trigger end_event explicitly. for example if the start and end event are derived from the same signal source.

LarsAsplund, 2017-06-07:
Sounds reasonable. Let me have a closer look next week. An alternative to creating the end event is also something like this

``` vhdl
    monitor: process is
      variable previous_data : std_logic_vector(data'range);
    begin
      previous_data := data;
      wait until rising_edge(clk);
      check_implication(data /= previous_data, start_event = '1');
    end process monitor;
```

cjchin, 2017-06-08:
Yes agreed that we can use edge detector in the manner that you described and then to trigger end_event. 

That is just the additional overhead that the test bench would incur and I am trying to aim for the cleanest way to represent each testbench. 

We could develop a bunch of ultilities to wrap certain functions in Vunit but I am afraid that eventually these tools will eventually become lost or forgotten and its value becomes diminished in spite of the efforts taken to develop and to maintain it. Therefore the best would be to build upon the ecosystem of a particular library or framework like VUnit itself and to share it with the community which may also contribute to it. I believe you mentioned about this possibility in another issue so i will comment more about it. I am very excited about the potential of it.

LarsAsplund, 2017-06-08:
@cjchin Agree, the suggested solution is just a workaround.  An improved `check_stable` would be a cleaner solution.

LarsAsplund, 2017-06-12:
@cjchin What would be the functionality of `end_event` is such a mode?
* Allowed to end the window before the next start event
* Any activity is ignored
* An end event is an error

LarsAsplund, 2017-09-12:
@cjchin I pushed an update for this. In case there is an end event in this mode it will still close the window.

closed 2017-09-12 by LarsAsplund