VUnit/vunit #250: set_sim_option before add_config has no effect

drizzd, 2017-06-07:
Hi,

If I call set_sim_option before add_config, then it has no effect:

    ui = VUnit.from_argv()
    ui.set_sim_option("modelsim.init_file.gui", "startup.do")
    testbench = lib.entity("tb_my_entity")
    testbench.add_config("default")
    ui.main()

It takes effect if I reorder like this:

    ui = VUnit.from_argv()
    testbench = lib.entity("tb_my_entity")
    testbench.add_config("default")
    ui.set_sim_option("modelsim.init_file.gui", "startup.do")
    ui.main()

Is this intentional?

kraigher, 2017-06-07:
Those two should be equivalent, however the add_source_file command needs to have happened before the set_sim_option else it does not "hit" any test bench because there is none yet. You have not shown an add_source_file command in your example, are you sure it is not that problem that is at play?

drizzd, 2017-06-09:
It is as you say. It is only the order of set_sim_option and add_source_file that matters.

kraigher, 2017-06-09:
That is the way it is intended.

drizzd, 2017-06-12:
Currently, doing this via `lib.set_sim_option` is an error. Would it be ok to make this an error for `ui.set_sim_option` as well, issuing an error message such as "error: No testbenches found. Set simulation options after adding testbenches"?

I for one ran into this issue without realizing what was going on. Is there a use case for setting simulation options when testbenches are not yet added?

    $ cat run.py
    from os.path import join, dirname
    from vunit import VUnit

    root = dirname(__file__)

    ui = VUnit.from_argv()
    lib = ui.add_library("lib")
    lib.set_sim_option("disable_ieee_warnings", True)
    lib.add_source_files(join(root, "*.vhd"))
    ui.main()

<div />

    $ python run.py 
    Traceback (most recent call last):
      File "run.py", line 14, in <module>
        lib.set_sim_option("disable_ieee_warnings", True)
      File "/local/bhr/vunit/vunit/ui.py", line 1020, in set_sim_option
        for test_bench in self.get_test_benches():
      File "/local/bhr/vunit/vunit/ui.py", line 1206, in get_test_benches
        for test_bench in self._test_bench_list.get_test_benches_in_library(self._library_name):
      File "/local/bhr/vunit/vunit/test_bench_list.py", line 50, in get_test_benches_in_library
        return self._libraries[library_name].values()
    KeyError: 'lib'

kraigher, 2017-07-19:
I have fixed the crash and added error for when there is no object matching the operation which can be overridden by `allow_empty` argument.

closed 2017-07-19 by kraigher