VUnit/vunit #251: Add simulator specific stuff to the modelsim.ini

gsorrenti, 2017-06-08:
I need to customize modelsim.ini. My gol is to set a flag (vhdlvariablelogging) sometimes for debug purpose. There is two way to give this option vsim parameter and modelsim.ini. It seems (and this is probably a modelsim bug) that the second way overwrite the first. Now I need to set the flag in modelsim.ini by hand but it is overwritten any time a exec a `run.py --clean`.
The result in modelsim.ini should be:
```
[vsim]
vhdlvariablelogging = 1
```
A more generic function to customize any aspect of modelsim.ini should be usefull because modelsim lacks in mapping function via TCL commands and so in those case the startup script is not a solution.

kraigher, 2017-06-10:
@gsorrenti Check out the branch issue_251, I added a `VUNIT_MODELSIM_INI` environment variable to point out the modelsim.ini file to use as the starting point instead of the installed modelsim.ini files which is the default. Please let me know if it solves your problem and we can make it an official solution. I am on a trip next week so I will not be that responsive then.

gsorrenti, 2017-06-16:
@kraigher It's work, thank you.

commit 42a54961 (2017-06-17): Adds option to set user defined modelsim.ini as starting point. Closes #251
closed 2017-06-17 by kraigher