VUnit/vunit #252: Proposition for Check_Next where num_clks => 0

cjchin, 2017-06-09:
Would it be possible to consider being able to set num_clks with parameter of 0? Meaning that clock starts that check_next check on the same cycle as the next_signal.  

Compilation fails when i try to set it so i think this is part of the design for this function but I find that there are times when i know that at this a particular instant, the signal will be asserted and not 1 clock cycle later (which is my limit at the moment). So then the test is not so clean because I have to always have to enable start_event one clock cycle before. 

Take for example if I wanted to make a timer function (derived from the next_Check function); to check that my signal is asserted for N number of clock cycles. The solution would not be elegant because I would always have to consider enabling the timer one clock before the signal itself and then stop the timer one clock before.

LarsAsplund, 2017-06-12:
@cjchin Compilation fails because `num_cks` is a `positive` so only values greater than zero are allowed. If you with `num_cks = 0` mean that `expr` must be `1` when `start_event` is `1` you're really expressing an implication that can be solved with `check_implication`

`check_next` looks at `expr` in one clock cycle as specified by `num_clk` and it's not intended to check if a signal is asserted for many cycles.

cjchin, 2017-06-13:
@LarsAsplund using check_next seems to be for me the only way to do what I am trying to do. 

For example, sometimes i have a std_logic signal which upon some event is asserted. I need to check the correctness of this signal, for example, the duration for how long this signal will be asserted for. 

When going through the check_api, it originally occured to me that the function that i was looking for was Check_Next, since by design, it will check that a signal is asserted for however long my start_event was asserted for. 

In this way, i can indirectly create the timer function which i described earlier. 

Do you have any suggestion for me to accomplish the task that I am trying to do? 

And anyway why not allow num_clks to have the possibility to be set as zero?

LarsAsplund, 2017-09-08:
@cjchin Ok, I understand your use case and added the requested support.

closed 2017-09-08 by LarsAsplund