VUnit/vunit #253: dictionary parameter for post_check( tcpath, generic_dic = None)

asicnet, 2017-06-09:
Hi
In the testbench configuration i can define the generics as a dictionary and a post_check script.
The Parameter for this script is only the path. It will be nice if a second parameter is given as a dictionary with all generics. Then an analyze is better possible.

    generics=dict( 
        g_use_busy   = busy,
        g_mode       = mode,
        g_reg        = reg,
        g_ports      = ports,
        g_data_bits  = bits,
        g_random     = random,
        g_frmln      = frmln
    ),
    post_check=post_check_data

def post_check_data(mpath):
    gtc = GenericTestAnalyze(mpath)
    res = gtc.check_data() 
    return res****

kraigher, 2017-06-09:
The post_check function does not really need that information since you already can know about it as a user. The only thing you cannot know is the output_path which is why it is provided. It is all quite easy in Python, consider the following template using nested functions to create a test case with pre_config and post_check functions:
```python
def create_config(name, test_bench, my_test_parameters):
    generics = f(my_test_parameters)
    
    def pre_config(output_path):
        # I can use my_test_parameters or generics here

    def post_check(output_path):
        # I can use my_test_parameters or generics here

    test_bench.add_config(name, generics=generics, post_check=post_check, pre_config=pre_config)

It is also possible to use classes and objects to keep the information as a user.
```

asicnet, 2017-06-10:
Have it tried.
Perfect!
Thank yoo

closed 2017-12-22 by kraigher